Klipper3d/klipper #5440: Input Shaper calibration crashes mcu when using AWD (4steppers on corexy) & 16 microsteps

MarvinBeym, 2022-04-21:
When trying to run `SHAPER_CALIBRATE` on a corexy machine with 4 steppers (x, x1 & y, y1)   
it will crash at `68hz` if the steppers are setup to use `16 microsteps`

## Error message
MCU 'mcu' shutdown: Timer too close
[klippy.log](https://github.com/Klipper3d/klipper/files/8533466/klippy.log)

## Steps to reproduce
1. Use CoreXY printer with AWD setup.
2. Change microsteps of all x/y steppers to `16`.
3. Run  `SHAPER_CALIBRATE`.
4. Crashes at `Testing frequency 68 Hz`.

## Tested
- Removing 2 steppers solves the issue.
- Changing all steppers to use 8 microsteps solves the issue  .
(at least for me, though I know someone else can't do it even with 8 microsteps).
- Going even higher with the microsteps will cause the crash to happen earlier.
- Reducing microsteps on other steppers like the Z-Axis (I have 3 Z Steppers) will still cause the issue at the exact same hz value.

## Setup

HevORT:
- AWD (4x LDO-57STH41-2804MAC(HEV))
- TMC2209 on all steppers
- FYSETC Spider v1.0
- Toolboard (irrelevant for the issue)

## Notes
- I'm now running a toolboard, the issue was already present before installing it, it has nothing to do with the issue.

MarvinBeym, 2022-04-22:
[image]

Sineos, 2022-04-23:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: [Klipper Discourse](https://klipper.discourse.group/)

MarvinBeym, 2022-04-23:
I don't see how this issue is not about development...

=> [Contact - I have diagnosed a defect in the Klipper software](https://www.klipper3d.org/Contact.html#i-have-diagnosed-a-defect-in-the-klipper-software)

To me it is clearly a bug report for multiple features not working together like they should.
All of these features are implemented and should have been tested to work.

In this case we are talking about
- AWD
- CoreXY (if it even matters, maybe this issue is present on all kinematic types)
- Input Shaper

RedFlash49, 2022-04-23:
This IS an bug of Klipper and not a feature/support request!

RedFlash49, 2022-04-23:
> As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: [Klipper Discourse](https://klipper.discourse.group/)
Open your eyes please!

closed 2022-04-25 by github-actions[bot]

Superchicken, 2022-04-25:
@Sineos please clarify why this doesn't fall under development, or reopen this issue. It appears to be an issue multiple users are experiencing with different setups/configurations.

Sineos, 2022-04-25:
https://www.klipper3d.org/Contact.html#i-have-diagnosed-a-defect-in-the-klipper-software --> Point 1

FWIW, I have already seen 4 stepper CoreXY machines perform it successfully. So it might be an issue or not.