Klipper3d/klipper #5451: error: Internal error in MCU 'mcu' stepcompress

GusBricker, 2022-04-23:
Sorry to open yet another one of these issues as I see there is a few, however after reading them all I am a bit stumped as this seems like it might be a bug as I am on a pretty recent version of Klipper `v0.10.0-319-g3e0c0e9c`. Was able to print a calibration cube fine but with the gcode provided I started running into this issue. Printer is a Geeetech A10 with BMG extruder, have attached klippy.log, printer.cfg and gcode. 

Happy to provide any further details.
[klippy.log](https://github.com/Klipper3d/klipper/files/8548749/klippy.log)
[printer.cfg.zip](https://github.com/Klipper3d/klipper/files/8548750/printer.cfg.zip)
[GA10_rpi-box.gcode.zip](https://github.com/Klipper3d/klipper/files/8548751/GA10_rpi-box.gcode.zip)

GusBricker, 2022-04-25:
@Sineos so you are confident this is not a bug?

jakep82, 2022-04-25:
From the [Klipper Contact page:](https://www.klipper3d.org/Contact.html#i-have-diagnosed-a-defect-in-the-klipper-software)

> Be sure the bug is in the Klipper software. If you are thinking "there is a problem, I can't figure out why, and therefore it is a Klipper bug", then do not open a github issue. In that case, someone interested and able will need to first research and diagnose the root cause of the problem. If you would like to share the results of your research or check if other users are experiencing similar issues then you can search the [Klipper Community Forum](https://www.klipper3d.org/Contact.html#community-forum).

closed 2022-04-26 by GusBricker