Klipper3d/klipper #5454: STM32F401 unable to flash generated firmware

anmarcu, 2022-04-24:
Hi
I'm tring to install klipper on the new Ender 3 S1 PRO.
The motherboard for this new printer use an STM32F401 which seems to be unable to be flashed with an sd card, also i've tried to flash it from USB tyoe C... Same...
Any ideea how i can flash the bin file generated by klipper?

Thanks

anmarcu, 2022-04-24:
[image]
[image]

This is the processor and these are the asettings.

Fisheiyy, 2022-04-24:
Im pretty sure you cant flash by USB but you should be able to via SD Card

anmarcu, 2022-04-24:
> Im pretty sure you cant flash by USB but you should be able to via SD Card, You can try to change the bootloader size.

The flash was done by SD Card. I used usb type c just to connect the printer to raspberry 3b+.
Also i've tried all 3 options for bootloaders.

Fisheiyy, 2022-04-25:
> > Im pretty sure you cant flash by USB but you should be able to via SD Card, You can try to change the bootloader size.
> 
> The flash was done by SD Card. I used usb type c just to connect the printer to raspberry 3b+. Also i've tried all 3 options for bootloaders.

are you renaming the .bin file to firmware.bin and then trying to flash it from the sd card?

anmarcu, 2022-04-25:
> are you renaming the .bin file to firmware.bin and then trying to flash it from the sd card?

Yes, I did it, same..

Fisheiyy, 2022-04-25:
You need a 64KiB bootloader for this specific STM32F401 someone has already made a PR to add the option but until then you can just edit the Kconfig file like the PR did and get the same result

anmarcu, 2022-04-25:
Indeed, JamesH fixed the problem on discord. Kudos

closed 2022-04-25 by anmarcu

shschong, 2022-04-29:
Hi, I'm having trouble with this too. I flashed my board, STM32F401, and I get the following on Mainsail:
MCU 'mcu' shutdown: Not a valid ADC pin
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Can you share your printer.cfg file? Thanks

Fisheiyy, 2022-04-29:
> Hi, I'm having trouble with this too. I flashed my board, STM32F401, and I get the following on Mainsail: MCU 'mcu' shutdown: Not a valid ADC pin Once the underlying issue is corrected, use the "FIRMWARE_RESTART" command to reset the firmware, reload the config, and restart the host software. Printer is shutdown
> 
> I'm using the printer.cfg file that was made for the STM32F103. Can you share your printer.cfg file? Thanks

Hey, you can test out my configuration in PR #5332, it will help us both as you can review my config

cln-io, 2022-06-12:
> JamesH

Could you possibly share this "fix" ?

Thanks

Fisheiyy, 2022-06-12:
> > Indeed, JamesH fixed the problem on discord.
> > I managed to flash the printer and now it's working.
> > Kudos
> 
> @anmarcu Could you possibly share this "fix" / what issues you were facing, and JamesH "fixed"?
> 
> Thanks

if  you read the thread you will see the issue was resolved by building klipper with a 64KiB bootloader for F401

cln-io, 2022-06-12:
This wasn’t entirely clear to me.
Apologies.

Kleszcz1337, 2022-09-23:
**SOLUTION**
A lot of You guys likely got here like me from google looking for `professional firmware don't want to flash on ender 3 s1`.

There is a solution from Reddit where the 120iq guy just put the .bin file into the folder named `STM32F4_UPDATE` and it's working like magic.

Happy printing! :)

Link:
https://www.reddit.com/r/Creality/comments/u0dyxl/comment/i5hlab0/?utm_source=share&utm_medium=web2x&context=3

Fisheiyy, 2022-09-23:
> **SOLUTION** A lot of You guys likely got here like me from google looking for `professional firmware don't want to flash on ender 3 s1`.
>
> There is a solution from Reddit where the 120iq guy just [...]

This fix has been known for some while and I have updated the S1 configs to reflect the fact you *may* need to use the folder, but may not.