Klipper3d/klipper #5455: DELTA_CALIBRATE enhanced - results off by a magnitude of ~2x.

bernhardberger, 2022-04-25:
```
stepper_a: position_endstop: 198.892274 angle: 209.690669 arm_length: 243.948717
stepper_b: position_endstop: 198.662360 angle: 329.960667 arm_length: 243.638442
stepper_c: position_endstop: 199.035346 angle: 90.000000 arm_length: 243.326816
delta_radius: 121.597122
```

`SAVE_CONFIG` results in (was hoping it was just a display issue):
```
#*# [printer]
#*# delta_radius = 121.597122
#*#
#*# [stepper_a]
#*# angle = 209.690669
#*# arm_length = 243.948717
#*# position_endstop = 198.892274
#*#
#*# [stepper_b]
#*# angle = 329.960667
#*# arm_length = 243.638442
#*# position_endstop = 198.662360
#*#
#*# [stepper_c]
#*# angle = 90.000000
#*# arm_length = 243.326816
#*# position_endstop = 199.035346

.....
#*# [delta_calibrate]
#*# height0 = -0.238
#*# height0_pos = 63601.000,63585.000,63591.000
#*# height1 = -0.238
#*# height1_pos = 81102.000,81101.000,53990.000
#*# height2 = -0.238
#*# height2_pos = 61499.000,92252.000,61495.000
#*# height3 = -0.238
#*# height3_pos = 54428.000,78122.000,78133.000
#*# height4 = -0.238
#*# height4_pos = 60983.000,60958.000,82553.000
#*# height5 = -0.238
#*# height5_pos = 75515.000,54938.000,75498.000
#*# height6 = -0.238
#*# height6_pos = 86970.000,61151.000,61153.000
#*# distance0 = 64.82
#*# distance0_pos1 = 126298.213,127524.143,127530.410
#*# distance0_pos2 = 116313.383,136146.928,136162.651
#*# distance1 = 64.9
#*# distance1_pos1 = 126728.748,126658.220,127965.718
#*# distance1_pos2 = 122597.498,122520.251,143642.515
#*# distance2 = 64.83
#*# distance2_pos1 = 127594.674,126228.168,127530.410
#*# distance2_pos2 = 136225.057,116243.341,136162.651
#*# distance3 = 64.82
#*# distance3_pos1 = 128030.101,126659.420,126664.482
#*# distance3_pos2 = 143706.898,122538.182,122534.876
#*# distance4 = 64.8
#*# distance4_pos1 = 127594.912,127525.352,126233.830
#*# distance4_pos2 = 136229.011,136167.047,116248.999
#*# distance5 = 64.77
#*# distance5_pos1 = 126728.983,127960.056,126664.482
#*# distance5_pos2 = 122601.022,143636.848,122534.876
#*# distance6 = 64.71
#*# distance6_pos1 = 117166.574,134355.669,135761.226
#*# distance6_pos2 = 123314.471,121985.332,142999.149
#*# distance7 = 64.87
#*# distance7_pos1 = 123493.992,122163.520,141724.114
#*# distance7_pos2 = 136559.727,116527.713,135097.490
#*# distance8 = 64.79
#*# distance8_pos1 = 135823.894,117097.198,134370.592
#*# distance8_pos2 = 143063.669,123255.217,121999.341
#*# distance9 = 64.75
#*# distance9_pos1 = 141788.632,123434.054,122176.962
#*# distance9_pos2 = 135163.831,136500.971,116532.817
#*# distance10 = 64.89
#*# distance10_pos1 = 134436.795,135764.287,117102.300
#*# distance10_pos2 = 122065.365,142992.789,123251.860
#*# distance11 = 64.79
#*# distance11_pos1 = 122242.875,141717.762,123431.269
#*# distance11_pos2 = 116597.091,135081.861,136497.198
```

Position Endstops are off by a magnitude of ~2 as are the arms. Values before were:
```
#*# [printer]
#*# delta_radius = 226.978419
#*#
#*# [stepper_a]
#*# angle = 210.009046
#*# arm_length = 440.470000
#*# position_endstop = 397.351468
#*#
#*# [stepper_b]
#*# angle = 329.953971
#*# arm_length = 440.470000
#*# position_endstop = 397.132576
#*#
#*# [stepper_c]
#*# angle = 90.000000
#*# arm_length = 440.470000
#*# position_endstop = 397.150270

...

#*# [delta_calibrate]
#*# height0 = -0.238
#*# height0_pos = 63601.000,63585.000,63591.000
#*# height1 = -0.238
#*# height1_pos = 81102.000,81101.000,53990.000
#*# height2 = -0.238
#*# height2_pos = 61499.000,92252.000,61495.000
#*# height3 = -0.238
#*# height3_pos = 54428.000,78122.000,78133.000
#*# height4 = -0.238
#*# height4_pos = 60983.000,60958.000,82553.000
#*# height5 = -0.238
#*# height5_pos = 75515.000,54938.000,75498.000
#*# height6 = -0.238
#*# height6_pos = 86970.000,61151.000,61153.000
```

Since I'm using 0.9 degree steppers I suspect that the calibration script doesn't account for this and still calculates with the default 200 steps per rotation from a 1.8deg motor?
[moonraker.log](https://github.com/Klipper3d/klipper/files/8554721/moonraker.log)
[klippy.log](https://github.com/Klipper3d/klipper/files/8554722/klippy.log)

Sineos, 2022-04-25:
Please also follow the direction given here https://www.klipper3d.org/Contact.html# and post your topic on https://klipper.discourse.group

bernhardberger, 2022-04-25:
> Please also follow the direction given here https://www.klipper3d.org/Contact.html# and post your topic on https://klipper.discourse.group

Why do I need to open a discourse discussion when reporting an actual bug (seems like Klipper uses the wrong base values for heights)? This is a reproducible **issue** and I followed exactly the guidelines written in the Contact section.

jakep82, 2022-04-25:
The vast majority of issues reported here turn out to be configuration or hardware problems, not actual bugs.  We use Discourse to triage these reports and help users so the developers aren't inundated with help and feature requests.  If a bug is identified and confirmed after discussion on Discourse, then an issue can be opened here.

bernhardberger, 2022-04-25:
> The vast majority of issues reported here turn out to be configuration or hardware problems, not actual bugs. We use Discourse to triage these reports and help users so the developers aren't [...]

Thank you for that explanation.  This is what imho should be written in the contact.html and also the discourse lacks a dedicated "issues" section then as it's absolutely not clear what the correct place would be to report a bug/issue in discourse.

[image]

Could you please point me to the correct spot on where to report the issue that I discovered? It's neither a General Discussion, nor Frontend, Features, Configs, Macros or Developers related.

I'm sorry for derailing this issue but at least I now understand what's the intention here - but it's not very user friendly (yet). If someone points me to the correct place on where to report this I will move my issue there.

jakep82, 2022-04-25:
General Discussion is the catch all.  I would start by asking if anyone else has experienced this problem.

Regarding the contact page, I think the directions are pretty well defined.  Point 1 at [this link](https://www.klipper3d.org/Contact.html#i-have-diagnosed-a-defect-in-the-klipper-software) is the most relevant to this discussion.

closed 2022-04-27 by github-actions[bot]