Klipper3d/klipper #5461: Failed to verify BLTouch probe is raised; retrying. SKR mini

litwakjjjr, 2022-04-26:
Hi,
I am VERY new and a noob to klipper and everything so please be patient with me:)

I have an Ender 5+ for over a year and decided to try klipper. The printer was 100% functional prior to Klipper install/upgrade so I am thinking wiring is correct. I am able to home X and Y but the BL touch is throwing an error, attached.

Can someone please assist? Thanks!

[klippy-2.log](https://github.com/Klipper3d/klipper/files/8563840/klippy-2.log)

Sineos, 2022-04-26:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: [Klipper Discourse](https://klipper.discourse.group/)

masterxq, 2022-04-26:
Possibly I can confirm this, have 3D-Touch. The warning will be displayed. But the behavior differs on my printer, it completely works for probing Z and bed leveling also works perfect.
I also think this came with a recent update as I did not change the Hardware. Did not start any research because there is no real problem.

It appears when the probe is used first time after restarting klipper, then never again. Possibly a state is initiated wrong or something...

If somebody wants to start research on this, I would happily supply more data.

closed 2022-04-28 by github-actions[bot]