Klipper3d/klipper #5462: [temperature_fan] is not working as expected

markxyx, 2022-04-28:
Hi All,

According to klipper doc (As below):

```
[temperature_fan]
Temperature-triggered cooling fans (one may define any number of sections with a "temperature_fan" prefix). A "temperature fan" is a fan that will be enabled whenever its associated sensor is above a set temperature. By default, a temperature_fan has a shutdown_speed equal to max_power.
```

Temperature fan suppose to turn on when the temperature higher than targeted temperature. After I set this up, the cooling fan turns on until the target temperature is met. Behave the same as a heater. My config is something like this:

```
[temperature_fan Chamber_fan]
pin: BP15
#max_power:
#shutdown_speed:
#cycle_time:
#hardware_pwm:
#kick_start_time:
#off_below:
#tachometer_pin:
#tachometer_ppr:
#tachometer_poll_interval:
#   See the "fan" section for a description of the above parameters.
sensor_type: PT1000
sensor_pin: BP10
control: PID                        **# There is no heater attached to chamber. Why would this be a must-have parameter？**
#max_delta:
min_temp: 0
max_temp: 100
#   See the "extruder" section for a description of the above parameters.
pid_Kp:  20
pid_Ki: 20
pid_Kd: 20
#   The proportional (pid_Kp), integral (pid_Ki), and derivative
#   (pid_Kd) settings for the PID feedback control system. Klipper
#   evaluates the PID settings with the following general formula:
#     fan_pwm = max_power - (Kp*e + Ki*integral(e) - Kd*derivative(e)) / 255
#   Where "e" is "target_temperature - measured_temperature" and
#   "fan_pwm" is the requested fan rate with 0.0 being full off and
#   1.0 being full on. The pid_Kp, pid_Ki, and pid_Kd parameters must
#   be provided when the PID control algorithm is enabled.
#pid_deriv_time: 2.0
#   A time value (in seconds) over which temperature measurements will
#   be smoothed when using the PID control algorithm. This may reduce
#   the impact of measurement noise. The default is 2 seconds.
target_temp: 40.0
#   A temperature (in Celsius) that will be the target temperature.
#   The default is 40 degrees.
#max_speed: 1.0
#   The fan speed (expressed as a value from 0.0 to 1.0) that the fan
#   will be set to when the sensor temperature exceeds the set value.
#   The default is 1.0.
#min_speed: 0.3
#   The minimum fan speed (expressed as a value from 0.0 to 1.0) that
#   the fan will be set to for PID temperature fans.
#   The default is 0.3.
#gcode_id:
#   If set, the temperature will be reported in M105 queries using the
#   given id. The default is to not report the temperature via M105.
```

Did I config it wrong? How do I make the fan behave like a cooling fan?

And now I set the chamber fan the same way as a extruder cooling fan. This way works, but I would need to change config file when I need the change the chamber temperature. Which is a bit annoying and I can't change it during printing. Hope anyone could figure out what I have been doing wrong.

nigelpjames, 2022-04-28:
You need a target temp for it to reach before enabling the fan - e.g. target_temp: 50.0

markxyx, 2022-04-28:
> arget_temp: 40.0

I set target temp to 40. The fan is on below 40 and off higher than 40.

nigelpjames, 2022-04-28:
I use control: watermark, not pid to manage the chamber fan and turn it on when it exceeds 50c

nigelpjames, 2022-04-28:
I use the following which maintains the chamber at 46c, ramping the chamber fan up and down to a max speed of 0.85 & off below 0.2. Pins and thermistor settings removed...

shutdown_speed:             0.0
kick_start_time:            0.5
off_below:                  0.19
min_temp:                   0
max_temp:                   55.0
target_temp:                46.0
max_speed:                  0.85
min_speed:                  0.0
control:                    pid
pid_Kp:                     2.0     
pid_Ki:                     5.0     
pid_Kd:                     0.5     
pid_deriv_time:             2.0
gcode_id: C

Sineos, 2022-04-29:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: [Klipper Discourse](https://klipper.discourse.group/)

closed 2022-04-29 by markxyx