Klipper3d/klipper #5465: SKR 1.4 Turbo Problem with spi_flash: reduce speed to 400K

TimoMF, 2022-04-28:
After this [spi_flash: reduce speed to 400K](https://github.com/Klipper3d/klipper/commit/dc28257a7547d282543d011dbe1e9b513a32a8ae) update I can no longer flash the firmware via SPI.

The error message:
```
Flashing /home/pi/klipper/out/klipper.bin to /dev/serial/by-id/usb-Klipper_lpc1769_0920010F27903CAF13A86D5CC72000F5-if00
Checking FatFS CFFI Build...
Connecting to MCU...Connected
Checking Current MCU Configuration...Done
MCU needs restart: is_config=1, is_shutdown=0
Attempting MCU Reset...Done
Waiting for device to reconnect...Done
Connecting to MCU...Connected
Initializing SD Card and Mounting file system...

SD Card Information:
Version: 2.0
SDHC/SDXC: False
Write Protected: False
Sectors: 245760
manufacturer_id: 0
oem_id: 0
product_name: APPSD
product_revision: 0.0
serial_number: 00000000
manufacturing_date: 12/2019
capacity: 120.0 MiB
fs_type: FAT16
volume_label:
volume_serial: 3399620564
Uploading Klipper Firmware to SD Card...Done
Validating Upload...
SD Card Flash Error: Error reading firmware.bin from SD
Traceback (most recent call last):
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1202, in main
    spiflash.run()
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1154, in run
    self.run_reactor_task(self.run_sdcard_upload)
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1138, in run_reactor_task
    k_reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 269, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 310, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/pi/klipper/klippy/reactor.py", line 156, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/pi/klipper/klippy/reactor.py", line 48, in invoke
    res = self.callback(eventtime)
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1118, in run_sdcard_upload
    self.firmware_checksum = self.mcu_conn.sdcard_upload()
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 978, in sdcard_upload
    raise SPIFlashError("Error reading %s from SD" % (fw_path))
SPIFlashError: Error reading firmware.bin from SD
```

If I put it back again, it goes again.
here's how it works again 'SD_SPI_SPEED = 4000000'

Greeting Timo

Sineos, 2022-05-01:
I con confirm @TimoMF findings.

Current git:
```
ubuntu@ubuntu ~/klipper $ git show -s
commit 722ad4a1d2798d600ad34c8c8e7d74171e018aef (HEAD -> master, origin/master, origin/HEAD)
Author: Kevin O'Connor <kevin@koconnor.net>
Date:   Sat Apr 30 14:51:14 2022 -0400

    docs: Update Example_Configs.md to note that baud should be 250000

    Signed-off-by: Kevin O'Connor <kevin@koconnor.net>
```

```
ubuntu@ubuntu ~/klipper $ ./scripts/flash-sdcard.sh /dev/ttyACM1 btt-skr-turbo-v1.4
Flashing /home/ubuntu/klipper/out/klipper.bin to /dev/ttyACM1
Checking FatFS CFFI Build...
Connecting to MCU...Connected
Checking Current MCU Configuration...Done
MCU needs restart: is_config=1, is_shutdown=0
Attempting MCU Reset...Done
Waiting for device to reconnect....Done
Connecting to MCU...Connected
Initializing SD Card and Mounting file system...
SD Card Flash Error: Failed to Initialize SD Card. Is it inserted?
Traceback (most recent call last):
  File "/home/ubuntu/klipper/scripts/spi_flash/spi_flash.py", line 1202, in main
    spiflash.run()
  File "/home/ubuntu/klipper/scripts/spi_flash/spi_flash.py", line 1154, in run
    self.run_reactor_task(self.run_sdcard_upload)
  File "/home/ubuntu/klipper/scripts/spi_flash/spi_flash.py", line 1138, in run_reactor_task
    k_reactor.run()
  File "/home/ubuntu/klipper/klippy/reactor.py", line 269, in run
    g_next.switch()
  File "/home/ubuntu/klipper/klippy/reactor.py", line 310, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/ubuntu/klipper/klippy/reactor.py", line 156, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/ubuntu/klipper/klippy/reactor.py", line 48, in invoke
    res = self.callback(eventtime)
  File "/home/ubuntu/klipper/scripts/spi_flash/spi_flash.py", line 1111, in run_sdcard_upload
    self.mcu_conn.configure_mcu(printfunc=output_line)
  File "/home/ubuntu/klipper/scripts/spi_flash/spi_flash.py", line 946, in configure_mcu
    "Failed to Initialize SD Card. Is it inserted?")
SPIFlashError: Failed to Initialize SD Card. Is it inserted?
```

Change SD_SPI_SPEED = 4000000
```
ubuntu@ubuntu ~/klipper $ nano ./scripts/spi_flash/spi_flash.py
ubuntu@ubuntu ~/klipper $ ./scripts/flash-sdcard.sh /dev/ttyACM1 btt-skr-turbo-v1.4
Flashing /home/ubuntu/klipper/out/klipper.bin to /dev/ttyACM1
Checking FatFS CFFI Build...
Connecting to MCU...Connected
Checking Current MCU Configuration...Done
MCU needs restart: is_config=1, is_shutdown=0
Attempting MCU Reset...Done
Waiting for device to reconnect....Done
Connecting to MCU...Connected
Initializing SD Card and Mounting file system...

SD Card Information:
Version: 2.0
SDHC/SDXC: True
Write Protected: False
Sectors: 7744512
manufacturer_id: 3
oem_id: SD
product_name: SS04G
product_revision: 8.128
serial_number: 0B2D4046
manufacturing_date: 5/2014
capacity: 3.7 GiB
fs_type: FAT32
volume_label:
volume_serial: 4062047058
Uploading Klipper Firmware to SD Card...Done
Validating Upload...Done
Firmware Upload Complete: firmware.bin, Size: 24544, Checksum (SHA1): 03FBF77B9432A16BBB526AEC66BA5950AEC604A6
Attempting MCU Reset...Done
Waiting for device to reconnect.....Done
Connecting to MCU...Connected
Verifying Flash...Version matched...Done
Firmware Flash Successful
Current Firmware: v0.10.0-391-g722ad4a1
Attempting MCU Reset...Done
SD Card Flash Complete
ubuntu@ubuntu ~/klipper $
```

ping @Arksine

Sineos, 2022-06-04:
Strange, working now. Reboot of the Pi fixed it? 🤷‍♂️ 
Continue on discourse if issue should pop up again

closed 2022-06-04 by Sineos