Klipper3d/klipper #5467: After Updating - Nothing works

SpaWn12, 2022-04-29:
I updated Klipper on my printer with mks 1.4 and TMC2130 

The comunication with spi didn't work.

I get the error:  Unable to write tmc spi 'stepper_x' register GCONF
[klippy.log](https://github.com/Klipper3d/klipper/files/8591388/klippy.log)

Nothing changed by the wiring, just updateding the firmware.

SpaWn12, 2022-04-29:
Now I get the error

TMC 'stepper_x' reports error: DRV_STATUS: 00000000 cs_actual=0(Reset?)
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
Klipper state: Not ready
[klippy.log](https://github.com/Klipper3d/klipper/files/8591985/klippy.log)

bill-orange, 2022-04-30:
Syntax looks odd in your config file for sensorless homing.  You have  driver_sgt = 20.  I think it should be something like driver_SGT: -64 

See if that makes a difference.

SpaWn12, 2022-04-30:
I started from scratch.
Now it’s working.

Sineos, 2022-05-01:
Closed as concluded. For the future please open such requests on https://klipper.discourse.group

closed 2022-05-01 by Sineos