Klipper3d/klipper #5472: Azteeg X5 Mini (Smoothieboard) not creating Serial port

AxMod3DPrint, 2022-05-01:
Hi,

I have just updated klipper to the latest pull. The board was working fine before updating. It seems the board isn't booting correctly and not creating the serial port. I have 2 printers connected to one Pi and the other is fine. Steps taken:

- Reflashed board for both lpc1769 & lpc1768 to check I have not flashed incorrectly, currently flashed for lcp1769 - both with the same outcome
- Checked `/dev/serial/by-id` no serial port appears for the board, only the other printer shows up: 
```pi@fluidd1:~/klipper $ ls /dev/serial/by-id/
usb-Klipper_stm32f103xe_32FFD9055359393542790743-if00
```
- Checked `lsusb` no port for the Azteeg shows up on both the pi and linux laptop:
```pi@fluidd1:~/klipper $ lsusb
Bus 001 Device 005: ID 045e:075d Microsoft Corp. LifeCam Cinema
Bus 001 Device 010: ID 1d50:614e OpenMoko, Inc. stm32f103xe
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
- Tried a spare Azteeg X5 Mini, Same result
- Changed cables, tried different ports, and on a Windows desktop and Linux laptop still the same issue
- Flashed Smoothieware to check the bootloader, both boards boot fine with smoothie

This makes me think that somewhere klipper isn't playing well with the smoothie bootloader. Klippy log attached - not sure it shows anything, with not being able to see the printer.  

[klippy-ax-02.log](https://github.com/Klipper3d/klipper/files/8599912/klippy-ax-02.log)

AxMod3DPrint, 2022-05-01:
I have also tested a Re-Arm I have on hand (LPC1768) and the same issue is happening with that as well

Sineos, 2022-05-02:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: [Klipper Discourse](https://klipper.discourse.group/)

AxMod3DPrint, 2022-05-02:
I would argue that this is a development issue. Klipper supported Smoothieware boards and somewhere from v0.10.0-1-gb806d71e -> v0.10.0-391 (taken form Fluidd) Klipper doesn't seem to be initialising the boot loader on LPC1768 & LPC1769 boards. This was working previously and is now not, so looks like a possible bug, hence posting here rather than on discourse. That having been said, I will post this there and see what response I get.

AxMod3DPrint, 2022-05-02:
Also issue #5475 seems to be a similar issue with the Spider board.

Sineos, 2022-05-02:
Please follow the directions at: [klipper3d.org/Contact.html](https://www.klipper3d.org/Contact.html)
It is the philosophy of this project to triage issues on Discord / Discourse. 99.873% of the users posting here argue that it is a development issue. Out of them maybe 1% really is one.

closed 2022-05-04 by github-actions[bot]