Klipper3d/klipper #5476: After last update (30.04.2022) i have problem wirth connection "error: mcu 'mcu': Serial connection closed"

meht41, 2022-05-02:
Recently updated my Klipper via Fluidd to the Actual Version v0.10.0-391-g722ad4a1 and after restart i have this error log :(

===========================================================
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/py/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/py/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/py/klipper/klippy/serialhdl.py", line 316, in get_response
    cmd_queue)
  File "/home/py/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/py/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
error: mcu 'mcu': Serial connection closed
===========================================================

Using:
Debian PC connected to MKS Robin Nano 1.2 via USB (serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0) baud: 250000
Klipper - v0.10.0-391-g722ad4a1
moonraker - v0.7.1-562-gcd520ba

Kiauh is detecting my MCU via USB. (/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0)

Done:
-Rebuild and Reflash MCU with new Klipper firmware. (No effect)
-Reinstall Klipper, Moonraker, Fluid. (No effect)

meht41, 2022-05-02:
[printer.zip](https://github.com/Klipper3d/klipper/files/8603989/printer.zip)

Sineos, 2022-05-02:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: [Klipper Discourse](https://klipper.discourse.group/)
Also attach your klippy.log there.

closed 2022-05-04 by github-actions[bot]