Klipper3d/klipper #5489: Internal error in stepcompress

myronschram, 2022-05-06:
About every other file I try to print fails with a "Internal error in stepcompress" error. This started with the last update. Here is the error:
`Stats 56745.1: gcodein=0 mcu: mcu_awake=0.003 mcu_task_avg=0.000022 mcu_task_stddev=0.000023 bytes_write=1512245 bytes_read=1032500 bytes_retransmit=1239 bytes_invalid=0 send_seq=36332 receive_seq=36329 retransmit_seq=36332 srtt=0.002 rttvar=0.000 rto=3.200 ready_bytes=192 stalled_bytes=554 freq=72001924 heater_bed: target=80 temp=80.0 pwm=0.208  sysload=0.08 cputime=987.718 memavail=3329672 print_time=7597.704 buffer_time=1.117 print_stall=6 extruder: target=285 temp=284.8 pwm=0.741
stepcompress o=2 i=0 c=22 a=0: Invalid sequence
Exception in flush_handler
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/toolhead.py", line 392, in _flush_handler
    self.flush_step_generation()
  File "/home/pi/klipper/klippy/toolhead.py", line 337, in flush_step_generation
    self.move_queue.flush()
  File "/home/pi/klipper/klippy/toolhead.py", line 175, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 333, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 286, in _update_move_time
    sg(sg_flush_time)
  File "/home/pi/klipper/klippy/stepper.py", line 226, in generate_steps
    raise error("Internal error in stepcompress")
error: Internal error in stepcompress
Stats 56746.1: gcodein=0 mcu: mcu_awake=0.003 mcu_task_avg=0.000022 mcu_task_stddev=0.000023 bytes_write=1512245 bytes_read=1032500 bytes_retransmit=1416 bytes_invalid=0 send_seq=36332 receive_seq=36329 retransmit_seq=36332 srtt=0.002 rttvar=0.000 rto=5.000 ready_bytes=192 stalled_bytes=555 freq=72001924 heater_bed: target=80 temp=80.0 pwm=0.208  sysload=0.08 `

Is there a work around? I have attached a link to the [Gcode ](https://drive.google.com/file/d/1DZGAR52GFXXH_lUCBMEQnU3zF5zCSrzC/view?usp=sharing)in case you need it.

Sineos, 2022-05-07:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

closed 2022-05-09 by github-actions[bot]

myronschram, 2022-05-09:
This is a coding issue

Thanks,
Myron