Klipper3d/klipper #5494: Bed Mesh levelling - Accept fail if z position is 0.000 or positive value.

yoodog, 2022-05-10:
Klipper latest is running  great, but when doing mesh bed levelling I encounter a problem 
- Manual probe failed! Use TESTZ commands to position the
// nozzle prior to running ACCEPT.

At first it seems to happen random. But after 4h of levelling I discovered that it happens ONLY when original point where it goes during starting sequence (defined and stored by Z_ENDSTOP_CALIBRATE, in my case it is at 0.000) is lower than the mesh point you are trying to save. OR SAME.

this effectively means
- no point on the entire bed can be same z_endstop hight
- no point on the entire bed can be above z_endstop hight

unless I miss understand z_endstop, I don't see why a point on mesh could not be above.
- my z_endstop is set where I have very slight drag on a paper = 0.1 above bed surface give or take (homming is 0.19 above that point) 

I can imagine why mesh could not be above homing high for example, but I don't see why it could not be equal to Z endstop.

so why is it not possible to accept Z position at 0.000 or positive value during bed mesh levelling ? Bag?

yoodog, 2022-05-10:
effectively this means that I have to have my homing corner (which is where Z_endstop is calibrated) to be THE highest location on the bed, so I have to intentionally bring one of the bed corners unnecessarily high, than mesh bed levelling spots will all be bellow Z_endstop point and mesh bed levelling calibration won't fail. 

that doesn't seem like correct operation

dewi-ny-je, 2022-05-10:
Have you tried setting "position_min" to -1 mm?

Sineos, 2022-05-11:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

Also attach your klippy.log file there

closed 2022-05-13 by github-actions[bot]