Klipper3d/klipper #5495: Dual Extruders on TMC2209's - Klipper cannot configure

munsterlander, 2022-05-10:
So I am trying to setup dual extruders that have a single hotend.  I have followed the docs, the forums, and discord and have multiple people stumped.  

Klipper docs say to do: https://www.klipper3d.org/Config_Reference.html#extruder_stepper
The suggested thread in the forums say: https://klipper.discourse.group/t/switching-between-extruders/2020/15

Which supports the docs, except with 2209's configured, you get interesting errors.

Here is the relevant config:

```
[extruder]
step_pin: PD15
dir_pin: !PD14
enable_pin: !PC7
microsteps: 16
rotation_distance: 33.500
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PB3
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA2
#control: pid
#pid_Kp: 22.2
#pid_Ki: 1.08
#pid_Kd: 114
min_temp: 0
max_temp: 350

[extruder_stepper belted_extruder]
extruder: extruder
step_pin: PD11
dir_pin: PD10
enable_pin: !PD13
microsteps: 16
rotation_distance: 33.500

[tmc2209 extruder]
uart_pin: PC6
run_current: 0.600
stealthchop_threshold: 999999

[tmc2209 belted_extruder]
uart_pin: PD12
run_current: 0.600
stealthchop_threshold: 999999

[gcode_macro T0]
gcode:
    # Deactivate stepper in my_extruder_stepper
    SYNC_EXTRUDER_MOTION EXTRUDER=belted_extruder MOTION_QUEUE=
    # Activate stepper in extruder
    SYNC_EXTRUDER_MOTION EXTRUDER=extruder MOTION_QUEUE=extruder
    #SET_PRESSURE_ADVANCE ADVANCE=0 EXTRUDER=belted_extruder
 
[gcode_macro T1]
gcode:
    # Deactivate stepper in extruder
    SYNC_EXTRUDER_MOTION EXTRUDER=extruder MOTION_QUEUE=
    # Activate stepper in my_extruder_stepper
    SYNC_EXTRUDER_MOTION EXTRUDER=belted_extruder MOTION_QUEUE=extruder 
    #SET_PRESSURE_ADVANCE ADVANCE=0 EXTRUDER=belted_extruder
```

When you save and restart, Klipper warns:  `Option 'microsteps' in section 'belted_extruder' must be specified`

That option is defined in the belted_extruder section but not in the 2209, so if you add it there, that error goes away and you get:  `rotation_distance must be specified.`  If you add that, then you get `Option 'rotation_distance' is not valid in section 'tmc2209 belted_extruder'`.  The only way this config runs is if I comment out the entire 2209 section for belted_extruder.

jakep82, 2022-05-10:
I believe you need to use the full name of the stepper. `[tmc2209 extruder_stepper belted_extruder]`

munsterlander, 2022-05-10:
Thank you. I will get this implemented asap.

closed 2022-05-10 by munsterlander