Klipper3d/klipper #5500: Unable To Start/Select Print After Update.

pinolt, 2022-05-13:
Fluidd
Klipper
Moonraker
OS Package 

Were updated on the 13th of May 2022, since the update I am unable to select a print regardless of the method : _click printer icon_ or _right click -> print_.

Please help me out,
Thanks
[klippy.log](https://github.com/Klipper3d/klipper/files/8683850/klippy.log)

dflorea4231, 2022-05-13:
This is what I see from the moonraker.log file

[moonraker (1).log](https://github.com/Klipper3d/klipper/files/8684325/moonraker.1.log)

```
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/websockets.py", line 253, in execute_method
    result = await callback(params)
  File "/home/pi/moonraker/moonraker/websockets.py", line 362, in func
    result = await callback(
  File "/home/pi/moonraker/moonraker/components/klippy_apis.py", line 75, in _gcode_start_print
    return await self.start_print(filename)
  File "/home/pi/moonraker/moonraker/components/klippy_apis.py", line 118, in start_print
    await self.klippy.wait_started()
  File "/home/pi/moonraker/moonraker/klippy_connection.py", line 115, in wait_started
    await asyncio.wait_for(
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```

hathawayoutthere, 2022-05-13:
[moonraker.log](https://github.com/Klipper3d/klipper/files/8684505/moonraker.log)

I am also having this issue after udpating Klipper/Moonraker an hour ago.

All commands work, but gcode uploads fail and any existing files cannot be printed.

I keep getting: "Internal Server Error"

teeedubb, 2022-05-13:
Can confirm this issue. Reverting to the prior commit solves it.

pinolt, 2022-05-13:
> Can confirm this issue. Reverting to the prior commit solves it.

How does one do this?

pinolt, 2022-05-13:
> I have this same issue. Just installed some new hardware today after being down for a couple of months and didn't know there was an update released today. Unable to print, but I can send all sorts of commands to the printer, as I was able to do my bed leveling and z offset, testing the extruder function and PID tunning. Everything works except for trying to actually send a job to print.
> 
> This is what I see from the moonraker.log file
> 
> [moonraker (1).log](https://github.com/Klipper3d/klipper/files/8684325/moonraker.1.log)
> 
> ```
> Traceback (most recent call last):
>   File "/home/pi/moonraker/moonraker/websockets.py", line 253, in execute_method
>     result = await callback(params)
>   File "/home/pi/moonraker/moonraker/websockets.py", line 362, in func
>     result = await callback(
>   File "/home/pi/moonraker/moonraker/components/klippy_apis.py", line 75, in _gcode_start_print
>     return await self.start_print(filename)
>   File "/home/pi/moonraker/moonraker/components/klippy_apis.py", line 118, in start_print
>     await self.klippy.wait_started()
>   File "/home/pi/moonraker/moonraker/klippy_connection.py", line 115, in wait_started
>     await asyncio.wait_for(
>   File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
>     raise exceptions.TimeoutError() from exc
> asyncio.exceptions.TimeoutError
> ```

Any ideas on how to resolve the issues experienced with moonraker?

Sineos, 2022-05-13:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

Also attach your klippy.log / moonraker.log file there

pinolt, 2022-05-13:
> As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)
> 
> Also attach your klippy.log / moonraker.log file there

Thank you

hathawayoutthere, 2022-05-13:
Here is a temporary fix. I found (performed when ssh'd into the pi)

cd moonraker

git reset --hard 3d9f5dc6d9d106973dc636b1512d0ce26720ce34 && git clean -f

sudo systemctl restart moonraker

Xilius, 2022-05-13:
> 

Thanks this helped!

Arksine, 2022-05-13:
Sorry guys, this was a bug I introduced in a recent commit.  It should be fixed now.

closed 2022-05-15 by github-actions[bot]