Klipper3d/klipper #5504: Add function for negative "unretract extra length" input.

poon-holder, 2022-05-14:
Have mercy for raising a potentially futile issue,  though some of us gnarly crackheds would like to introduce a negative figure in the "unretract extra length" param. 
Is this a dirty practise, going in an opposite direction for Lord O'conner's most concise outlook on FDM tuning? If so, we will happily sort our lives out and conform to his standards.
Many Thanks, God bless.

Sineos, 2022-05-15:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

poon-holder, 2022-05-16:
Apologies sacred ones. I'm classically smooth brained.

closed 2022-05-16 by poon-holder