Klipper3d/klipper #5505: Getting various TMC issues since adding a 12864mini LCD

Rudis1261, 2022-05-14:
I have no idea where to start debugging this 🙈  

My printer setup is:
- Annex K3
- BTT Octopus 8 * TMC2209s
- 300w 24V PSU
- BTT 12864 v1.0 with the reset resistor removed (it was resetting the printer all the time)

I added the BTT 12864 LCD to my printer, and ever since I have been getting various TMC driver issues (TMC2209s sensorless homing). The printer performs 100% fine, but as soon as I add the `[display]` block, it becomes very unstable, and randomly gives one of many TMC fault codes. 

The config I got from all the BTT repos, and should be fine, and I have double-checked them against a Voron 2.4 config, which seem to be correct, except the neopixel index order, but that should be fine.

Rudis1261, 2022-05-14:
Any ideas/thoughts on how to isolate/trace the issue would be welcome!

Sineos, 2022-05-14:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

Also attach your klippy.log file there

See:
https://klipper.discourse.group/t/tmc-extruder-reports-error-gstat-00000001-reset-1-reset/1156/39?u=sineos

Rudis1261, 2022-05-14:
Cool cool, tnx 😢

closed 2022-05-14 by Rudis1261