Klipper3d/klipper #5509: Troubles with heating chamber. M191 implementation.

Octopusmode, 2022-05-17:
I'm trying to implement camera heating.
I have already set up the heaters and the cooling of the heating heater. The chamber heating works well and stable.
I have implemented the macro command M141 S<temp> T<heater_temp> "Turn on the chamber heating"
I can't implement command M191 "Turn on chamber heating and wait for set temperature" with the same parameters.
I try to use TEMPERATURE_WAIT, but SENSOR=<config_name> accepts neither [generic_heater] nor [temperature_sensor], but it accepts [extruder] for example.

_[gcode_macro M191]
gcode:
  {% set S = params.S | default(0) | float %}
  {% set T = params.T | default(110) | float %}
  SET_TEMPERATURE_FAN_TARGET TEMPERATURE_FAN=chamber_fan TARGET={T}
  M118 Chamber heating to {S}C
  SET_HEATER_TEMPERATURE HEATER=heater_chamber TARGET={S}
  M118 Chamber heater setting is {T}C
  M118 Waiting for chamber heating...
  TEMPERATURE_WAIT SENSOR=heater_chamber MINIMUM={S} 
  M118 Chamber heating {S}C is done._

**Macro result:**
_$ M191 S53
21:12:57 echo: Chamber heating to 53.0C
21:12:57 echo: Chamber heater setting is 110.0C
21:12:57 echo: Waiting for chamber heating...
21:12:57 !! Unknown sensor 'heater_chamber'
21:12:57 !! Unknown sensor 'heater_chamber'_

An on the web search for an answer was inconclusive.
How can I implement my task without changing the sources of the Klipper?

Sineos, 2022-05-18:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

Also attach your klippy.log then.

closed 2022-05-20 by github-actions[bot]

JRHeaton, 2022-05-31:
The reason for this problem is that Klipper registers the sensor names with the full, literal config section name. I provide a workaround at the end of this post. 

Here is a macro I created that prints out the available heaters and sensors:
```yaml
[gcode_macro list_sensors]
gcode:
  { action_respond_info(printer.heaters.available_heaters | join(', ')) }
  { action_respond_info(printer.heaters.available_sensors | join(', ')) }
```
And the output from my Mainsail instance:
[image]

This is how I found out about the literal name registration. It's not very ideal from a discovery or intuitive UX perspective, so perhaps we can petition @KevinOConnor to consider registering `heater_generic` and `temperature_sensor` sensor names with the name of their respective heaters. This would make the pairing of `SET_HEATER_TEMPERATURE` and `TEMPERATURE_WAIT` commands symmetric in their fields addressing the heater/sensor. 

**The workaround** (use the exact names in quotes for `SENSOR`):
`TEMPERATURE_WAIT SENSOR="heater_generic heater_chamber" ...`

Cheers.

psiberfunk, 2022-05-31:
> The reason for this problem is that Klipper registers the sensor names with the full, literal config section name. I provide a workaround at the end of this post.
>
> Here is a macro I created that [...]

Thanks so much for helping me solve this @JRHeaton !!