Klipper3d/klipper #5510: New slider (PA, smooth time) does not work with extruder_stepper config

homasholz, 2022-05-18:
The new sliders do not work with my extruder config. See screenshots attached. 

[image]
[image]
[image]
[image]
[image]

Sineos, 2022-05-18:
This is not a Klipper issue but an issue of the web interface (fluidd / mainsail) you are using.

closed 2022-05-18 by Sineos

homasholz, 2022-05-18:
oh yes, sorry was on wrong page 🙈