Klipper3d/klipper #5515: MCU 'mcu' shutdown: Scheduled pwm event will exceed max_duration               laser pwm issues

bihailantian655, 2022-05-18:
I'm using an  lightburn exported marlin gcode  file（m3 m4 ）,  After running for about 5 seconds, an error is reported

MCU 'mcu' shutdown: Scheduled pwm event will exceed max_duration

jakep82, 2022-05-19:
Lasers are not supported by Klipper. 

#4128

bihailantian655, 2022-05-20:
The issue is solved
Just turn this parameter up    cycle_Time

[output_pin TOOL]
pin: PH6    # use your fan's pin number
pwm: True
hardware_pwm: True
cycle_time: 0.1
shutdown_value: 0
maximum_mcu_duration: 5

closed 2022-05-20 by bihailantian655

bihailantian655, 2022-05-23:
1    machine  cpu   mega2560 .    
2    my firmware is   compiled with the master   branch，  

                          [output_pin TOOL]
                          pin: PH6    # use your fan's pin number
                          pwm: True
                          hardware_pwm: True
                          cycle_time: 0.1
                          shutdown_value: 0
                          maximum_mcu_duration: 5
                          # Default: 0 (disabled)
                          # Amount of time in which the host has to acknowledge
                          # a non-shutdown output value.
                          # Suggested value is around 5 seconds.
                          # Use a value that does not burn up your stock.
                          # Please note that during homing, your tool
                          # needs to be in default speed.
                          
                          [gcode_macro M3]
                          gcode:
                              {% set S = params.S|default(0.0)|float %}
                              SET_PIN PIN=TOOL VALUE={S / 255.0}
                          
                          [gcode_macro M4]
                          gcode:
                              {% set S = params.S|default(0.0)|float %}
                              SET_PIN PIN=TOOL VALUE={S / 255.0}
                          
                          [gcode_macro M5]
                          gcode:
                              SET_PIN PIN=TOOL VALUE=0

#  pwm examples   00

4   Some gcode will no longer  reporte error     If  set  cycle_time:  to  0.1   （not  all  ）

4    the    maximum value  of   maximum_mcu_duratio can only be set  to  5

Cirromulus, 2022-05-23:
You can disable the `maximum_mcu_duration` if you leave out this config element. But be aware that this disables this safety feature.
What was your `cycle_time` previously, before setting it to the default value of `0.1`?

bihailantian655, 2022-05-23:
thaks for  your advice 
the  previous   cycle_time:  0.001   ,