Klipper3d/klipper #5516: adc_voltage not allowed in adc_temperature section

vavaroutsos, 2022-05-18:
Documentation here says adc_voltage is allowed in an adc_temperature section.

https://www.klipper3d.org/Config_Reference.html#adc_temperature

However, I get the following error when I try it.

Option 'adc_voltage' is not valid in section 'adc_temperature btt_octopus_pt100'

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

[image]

Here is the relevant section of printer.cfg:

[adc_temperature btt_octopus_pt100]
adc_voltage: 3.3
#temperature1: 0
#voltage1: 0.7332
#temperature2: 500
#voltage2: 1.980494
temperature1: 0
voltage1: 0
temperature2: 3300
voltage2: 3.3

JamesH1978, 2022-05-19:
According to official documentation adc_voltage: is indeed not valid in an adc_temperature section, the description states "**A heater section using this sensor** may also specify adc_voltage and voltage_offset parameters to define the ADC voltage". So this option goes in the heater section that references this sensor, not the adc_temperature section itself.

Thanks
James

vavaroutsos, 2022-05-19:
James, thanks for the reply. Doesn’t that seem a bit strange to define the ADC voltage in a heater section? It seems much cleaner to define all of the ADC properties in the adc_temperature section. Maybe this could be taken under consideration.

 

In my particular case, I didn’t even have an associated heater. I was trying to debug some faulty INA826/PT100 sensor and I wanted to see what ADC input voltage was, so I configured the temperature sensor to display mV, but was unable to specify the ADC voltage.

 

Best Regards,

 

Pete

JamesH1978, 2022-05-19:
Hi Pete, I'm not a dev, so cant comment on the why, but that would not be a discussion for here that would be the discourse or the discord. We try to only use the GitHub issues for actual identified bugs. I'm sure someone with more rights on here will be along to tag this as not on GitHub issue.

Thanks

James

closed 2022-05-22 by github-actions[bot]