Klipper3d/klipper #5521: SKR2: MCU 'mcu' shutdown: oids already allocated

ConstantijnCrijnen, 2022-05-22:
I'm trying to replace an SKR1.4 for an SKR2, but I can't seem to be able to connect. I used the [generic SKR2 config](https://github.com/Klipper3d/klipper/blob/master/config/generic-bigtreetech-skr-2.cfg) with only the mcu serial path adjusted. 

The specific SKR2 board I got is the one with the stm32f407 and the correct HY1904 mosfet (older versions had a faulty driver protection mosfet).

Maybe someone could help me out, as I'm a bit lost on what to do at this point.

[klippy.log](https://github.com/Klipper3d/klipper/files/8750091/klippy-3.log)

ConstantijnCrijnen, 2022-05-22:
So the reason why I'm swapping the SKR1.4 for an SKR2 is because I dropped a crimp connector between the 24V and 3.3V rails of the SKR1.4 while it was on. It just came to me that I might have killed the USB ports on the Raspberry Pi as well, so I swapped in a new Pi and now it works. 

I don't understand what the broken ports have to do with object ids, but unless this turns out to be a fluke after all, I'd consider this thread to be solved.

Sineos, 2022-05-23:
If there is anything to update please post on [Klipper Discourse](https://klipper.discourse.group/)