Klipper3d/klipper #5523: python-dev isn't available on some debien distros now

charlespick, 2022-05-22:
Not aware if this has trickled down to raspian/raspberrypios, and older versions of Ubuntu but on latest debien from yesterday and Ubuntu 22, you get this error when running the instal-debian script: 

[image]
python-dev has no installation candidate

I replaced the dependancy with python2-dev and everything seems to be working. Should I open a pull to make it standard or is there a known problem with that or are we just holding off until the python3 migration happens?

charlespick, 2022-05-23:
annnnd apparently libusb isn't working either, something i noticed when trying to flash my new huvud boards. libusb-1.0-0-dev worked

voland696, 2022-05-27:
pi@xxxx:~ $ ls /dev/serial/by-id/
ls: cannot access '/dev/serial/by-id/': No such file or directory
pi@xxxx:~ $

On multiple machines

JamesH1978, 2022-05-29:
@charlespick this is known issue that ubuntu 22 has changed the names of some packages, feel free to submit a PR for a new install file, but i think the OG files should stay as they are, as the ubuntu install file clearly states 18.04, the debian one would have to be decision made by others.

@voland696 this is nothing to do with this issue, that is an error with flash or board connection to the host, and not an issue for github, please use the discord or discourse servers for this

Thanks
James

charlespick, 2022-05-29:
@JamesH1978 I see

libusb was having issues on moonrakeros though... think that's a personal problem with my install or raspberrypios and it's parents?

voland696, 2022-06-02:
> @charlespick this is known issue that ubuntu 22 has changed the names of some packages, feel free to submit a PR for a new install file, but i think the OG files should stay as they are, as the [...]

What makes you think so? Let me give you some more details please:
My MCU is connected by UART, USB camera, I have multiple machines with this setup and I just did a fresh, from scratch setup on one of the machines with mainsailOS with the same results.
I'm using BTT SKR2 and RPi3 or RPi4.
Everything works very well, BUT! If I use 
ls /dev/serial/by-id/
I'm getting the error message:
pi@xxx:~ $ ls /dev/serial/by-id/
ls: cannot access '/dev/serial/by-id/': No such file or directory

I checked files on RPi and /dev/serial/ is not there.
As I mentioned before, the USB cam is plugged in and works so it is supposed to show something but it's not.

lsusb output:
pi@xxx:~ $ lsusb
Bus 001 Device 006: ID 05a3:9331 ARC International Camera
Bus 001 Device 004: ID 1d50:606f OpenMoko, Inc.
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Other outputs that could be helpful:
pi@xxx:~ $ ls -l /dev/serial/by-path/
ls: cannot access '/dev/serial/by-path/': No such file or directory
pi@xxx:~ $ ls -l /dev/ttyUSB*
ls: cannot access '/dev/ttyUSB*': No such file or directory

pi@xxx:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pi@xxx:~ $ ls -al /dev/tty*
crw-rw-rw- 1 root tty       5,  0 Jun  1 23:46 /dev/tty
crw--w---- 1 root tty       4,  0 Jun  1 23:21 /dev/tty0
crw--w---- 1 root tty       4,  1 Jun  1 23:21 /dev/tty1
crw--w---- 1 root tty       4, 10 Jun  1 23:21 /dev/tty10
crw--w---- 1 root tty       4, 11 Jun  1 23:21 /dev/tty11
crw--w---- 1 root tty       4, 12 Jun  1 23:21 /dev/tty12
crw--w---- 1 root tty       4, 13 Jun  1 23:21 /dev/tty13
crw--w---- 1 root tty       4, 14 Jun  1 23:21 /dev/tty14
crw--w---- 1 root tty       4, 15 Jun  1 23:21 /dev/tty15
crw--w---- 1 root tty       4, 16 Jun  1 23:21 /dev/tty16
crw--w---- 1 root tty       4, 17 Jun  1 23:21 /dev/tty17
crw--w---- 1 root tty       4, 18 Jun  1 23:21 /dev/tty18
crw--w---- 1 root tty       4, 19 Jun  1 23:21 /dev/tty19
crw--w---- 1 root tty       4,  2 Jun  1 23:21 /dev/tty2
crw--w---- 1 root tty       4, 20 Jun  1 23:21 /dev/tty20
crw--w---- 1 root tty       4, 21 Jun  1 23:21 /dev/tty21
crw--w---- 1 root tty       4, 22 Jun  1 23:21 /dev/tty22
crw--w---- 1 root tty       4, 23 Jun  1 23:21 /dev/tty23
crw--w---- 1 root tty       4, 24 Jun  1 23:21 /dev/tty24
crw--w---- 1 root tty       4, 25 Jun  1 23:21 /dev/tty25
crw--w---- 1 root tty       4, 26 Jun  1 23:21 /dev/tty26
crw--w---- 1 root tty       4, 27 Jun  1 23:21 /dev/tty27
crw--w---- 1 root tty       4, 28 Jun  1 23:21 /dev/tty28
crw--w---- 1 root tty       4, 29 Jun  1 23:21 /dev/tty29
crw--w---- 1 root tty       4,  3 Jun  1 23:21 /dev/tty3
crw--w---- 1 root tty       4, 30 Jun  1 23:21 /dev/tty30
crw--w---- 1 root tty       4, 31 Jun  1 23:21 /dev/tty31
crw--w---- 1 root tty       4, 32 Jun  1 23:21 /dev/tty32
[... 24 lines cut ...]
crw--w---- 1 root tty       4, 55 Jun  1 23:21 /dev/tty55
crw--w---- 1 root tty       4, 56 Jun  1 23:21 /dev/tty56
crw--w---- 1 root tty       4, 57 Jun  1 23:21 /dev/tty57
crw--w---- 1 root tty       4, 58 Jun  1 23:21 /dev/tty58
crw--w---- 1 root tty       4, 59 Jun  1 23:21 /dev/tty59
crw--w---- 1 root tty       4,  6 Jun  1 23:21 /dev/tty6
crw--w---- 1 root tty       4, 60 Jun  1 23:21 /dev/tty60
crw--w---- 1 root tty       4, 61 Jun  1 23:21 /dev/tty61
crw--w---- 1 root tty       4, 62 Jun  1 23:21 /dev/tty62
crw--w---- 1 root tty       4, 63 Jun  1 23:21 /dev/tty63
crw--w---- 1 root tty       4,  7 Jun  1 23:21 /dev/tty7
crw--w---- 1 root tty       4,  8 Jun  1 23:21 /dev/tty8
crw--w---- 1 root tty       4,  9 Jun  1 23:21 /dev/tty9
crw-rw---- 1 root dialout 204, 64 Jun  2 23:25 /dev/ttyAMA0
crw------- 1 root root      5,  3 Jun  1 23:21 /dev/ttyprintk
crw-rw---- 1 root dialout   4, 64 Jun  1 23:21 /dev/ttyS0

pi@xxx:~ $ ls -l /lib/modules
total 16
drwxr-xr-x 3 root root 4096 Jun  1 20:07 5.10.103+
drwxr-xr-x 3 root root 4096 Jun  1 20:07 5.10.103-v7+
drwxr-xr-x 3 root root 4096 Jun  1 20:07 5.10.103-v7l+
drwxr-xr-x 3 root root 4096 Jun  1 20:07 5.10.103-v8+

What is it I'm missing here?

Sineos, 2022-06-03:
@charlespick 
@JamesH1978 is correct. A PR would be needed to create new `install-ubuntu-22.04.sh` to reflect these changes

@voland696 
Sorry, but your help request is indeed totally misplaced here. Please open a topic on https://klipper.discourse.group

closed 2022-06-03 by Sineos