Klipper3d/klipper #5536: klipper gcode macro names ending in a number won't accept = in parameter value

vavaroutsos, 2022-05-31:
I'm using the Mainsail GUI. The macro buttons pass parameter using "=" even if the macro name ends in a number. If I define a gcode macro like M141 S<temp>, Klipper doesn't properly handle calling the macro like this:

M141 S=45

This results in the value of S being 0. Calling like this:

M141 S45

results in S having the expected value of 45. I don't know if this is a Klipper of Mainsail issue, but it seems like Klipper should still accept an "=" for macros names that end in a number.

jakep82, 2022-06-01:
I can't comment on Mainsail, but this is intended behavior on the Klipper side.  See the link for an explanation. 

https://github.com/Klipper3d/klipper/issues/5111#issuecomment-1006039786

vavaroutsos, 2022-06-01:
Jake, thanks for the reply. I understand the reason for the behavior, and I’m not suggesting to get rid of it. What I’m suggesting is that for macros that end in a number, parameter are parsed either way. For example, both of the following would be valid:

 

G141 S=55                           ; This currently doesn’t work.

 

G141 S55

 

I can’t see how this could hurt anything. Macro buttons in Mainsail always pass parameter using an “=”, so trying to activate the above macro from the button doesn’t work.

 

Regards,

 

Pete

 

#5111 (comment) <https://github.com/Klipper3d/klipper/issues/5111#issuecomment-1006039786>

closed 2022-06-03 by github-actions[bot]