Klipper3d/klipper #5537: Support for GD32F303 in new creality boards

Kaminariss, 2022-05-31:
Hello, I am currently writing support for these new chips as it appears they are becoming more and more common (not only creality but also bigtreetech). I have this board on Sovol SV01 Pro.

What I have done already:
1. Included STM Standard library headers for STM32F30x (without their functions)
2. Wrote Kconfig that allows you to select F303 microcontroller
3. Rewrote part of the code until it compiles without error

Work can be found here (**it is still work in progress not ready for PR but i need some help with it**)
https://github.com/Kaminariss/klipper/commit/9fde7f7860a7d55bdce23cc6dec9ff1a41bc75d5

What is left

1. There are some big differences in registers that I could use some help with, noticeably with ADC/Serial and I2C
2. There are still some blanks left where registers did not exist and I was not sure what original code was suppose to do.
3. Reformat/reorganize code
4. Test it

Sineos, 2022-06-04:
I'd recommend to issue a PR to get knowledgeable developers to have a look.

pwilkowski, 2022-06-06:
all that work can be put to trash, apparently GD32F303 is simply compatible with STM32F103, yes i know, should not be possible but im running klipper with it no problems.

kaidegit, 2022-06-06:
Acually, GD32F303 is not a clone of STM32F303. It is still a compatible chip of STM32F103. The firmware and footprints are compatible. What's more, GD32F303 uses Cortex-M4F core instead of CM3 which STM32F1xx use. And CM3 is compatible with CM4. The perfomance are better, however it may not be important in klipper.

[Here](https://blog.csdn.net/qq_23852045/article/details/109802955) is an artical written by an FAE of GigaDevice., which expounded using GD32F103/E103/F303 to instead of STM32F103.

closed 2022-06-13 by KevinOConnor