Klipper3d/klipper #5541: No UUID provided on request -- CANBoot

Mosher23, 2022-06-01:
Hello, 

Issue:
I do not receive any UUID running `python3 flash_can.py -q`  only following message 
> pi@mainsail:~/klipper/lib/canboot $ python3 flash_can.py -q
Resetting all bootloader node IDs...
Checking for canboot nodes...
Query Complete

This is my second CAN-Board, which is EBB42 from BTT. First board was SHT42 from Mellow which I could get UUID running the command above.

Both boards where flashed with up-to-date CANBoot and flashed via dfu-util with success.

What I have tried:
-  stopping klipper then running the command
-  differenet speeds 250k and 500k ( of coarse reflashing suitable firmware and changing settings on RPi side)

Arksine, 2022-06-01:
If Klipper has assigned the node an ID then it will not return a UUID when queried.  You must reset the board in order to query the UUID.  FWIW, the UUID will not change, its the same for CanBoot as it is for Klipper.

If the device has not been flashed with Klipper and you are unable query a UUID then it suggests that it has not entered the bootloader.  If possible I suggest building CanBoot with both the led indicator enabled (flashing LED when the bootloader is active) and the "doubletap reset" enabled.   Also its a good idea to do a full chip erase before flashing CanBoot.

jakep82, 2022-06-01:
The schematic for the V1.0 says the LED is on PA14, but it also lists the MCU as a F042 instead of a F072.  I'll be flashing my boards with CanBoot today, so I'll see if that is the correct pin for the LED.

https://github.com/bigtreetech/EBB/blob/master/EBB%20CAN%20V1.0%20(STM32F072)/EBB42%20CAN%20V1.0/Hardware/EBB42%20CAN%20V1.0-SCH.pdf

jakep82, 2022-06-01:
I've confirmed that PA14 is the correct pin for the LED on a V1.0 EBB42.  After flashing CanBoot compiled with LED support, I have a blinking green LED as expected.

jakep82, 2022-06-02:
Works fine for me flashing over USB with the board in DFU mode using STM32CubeProgrammer.

Arksine, 2022-06-02:
I agree, it should be possible to upload CanBoot via DFU.  Make sure your CanBoot build has the `Support bootloader entry on rapid double click of reset button` enabled and configure the LED as Jake mentioned.  Then upload CanBoot and quickly press the reset button twice.  The led should start blinking, if so you know you are in business.

jakep82, 2022-06-03:
I flashed another EBB42 fresh out of the box using dfu-util. The mass-erase option is required, but it works fine with that flag. CanBoot is running as expected after using this method. 

`dfu-util -d 0483:df11 -a 0 -R -D canboot.bin -s0x08000000:mass-erase:force:leave`

jakep82, 2022-06-05:
Any reason this issue hasn't been closed?  I've provided a verified working solution, and it's not actually a Klipper issue anyways.

closed 2022-06-05 by Mosher23