Klipper3d/klipper #5557: Klipper cannot reconnect to the MCU after firmware restart since 8b1e3c3fb25e3d3fbec49158ba9dba9b07c9be00

compihu, 2022-06-08:
Stating with the commit aforementioned above, Klipper is not able to reconnect to the (SKR v1.3) MCU following a firmware reset.
Building and flashing the MCU firmware is enough to reproduce the issue, Klippy version does not seem to make difference.
On power-on, the connection builds up without problems.

Relevant klippy.log section:

```
mcu 'mcu': Starting serial connect
webhooks client 3036409280: New connection
webhooks client 3036409280: Client info {'program': 'Moonraker', 'version': '?'}
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/usr/lib/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/usr/lib/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/usr/lib/klipper/klippy/serialhdl.py", line 315, in get_response
    self.serial.raw_send_wait_ack(cmds[-1], minclock, reqclock,
  File "/usr/lib/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/usr/lib/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
serialhdl.error: mcu 'mcu': Serial connection closed
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/usr/lib/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/usr/lib/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/usr/lib/klipper/klippy/serialhdl.py", line 315, in get_response
    self.serial.raw_send_wait_ack(cmds[-1], minclock, reqclock,
  File "/usr/lib/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/usr/lib/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
serialhdl.error: mcu 'mcu': Serial connection closed
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
```

Relevant config section:

```
[mcu]
restart_method: command
serial: dev/ttyS0

```

Sineos, 2022-06-08:
As already [stated when you opened the ticket](https://www.klipper3d.org/Contact.html): Issues here are for development only. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

Also attach your klippy.log there

compihu, 2022-06-08:
OK, guys, your code your rules. If you are not interested in a regression in a specified commit I will not be the one who ruins your game.

Sineos, 2022-06-08:
Everybody posting here is absolute sure to have found a bug / regression, whatever. In around 99% cases it is a user problem. So it is common practice here to triage the issues via Discourse / Discord before deciding.

If you already know the offending line of code then it is something different and by all means go ahead and post it.

compihu, 2022-06-08:
@Arksine Thank, works fine.

closed 2022-06-10 by github-actions[bot]