Klipper3d/klipper #5567: DELTA_ANALYZE problem with Python 3

CODeRUS, 2022-06-12:
Extended delta calibration on Python 3:
```
stepper_a: position_endstop: 130.819094 angle: 300.291392 arm_length: 1620.103986
stepper_b: position_endstop: 779.127972 angle: 415.945147 arm_length: 773.217936
stepper_c: position_endstop: 637.737578 angle: 90.000000 arm_length: 830.716119
delta_radius: 727.540055
```

Python 2:
```
stepper_a: position_endstop: 311.189999 angle: 210.134550 arm_length: 269.294402
stepper_b: position_endstop: 310.941165 angle: 330.062081 arm_length: 269.087290
stepper_c: position_endstop: 311.407977 angle: 90.000000 arm_length: 268.545012
delta_radius: 135.710389
```

Commands i've used:
```
DELTA_ANALYZE CENTER_DISTS=73.94,74.28,73.94,74.13,74.16,73.90
DELTA_ANALYZE OUTER_DISTS=73.15,74.26,74.16,74.10,73.88,74.41
DELTA_ANALYZE CENTER_PILLAR_WIDTHS=8.89,8.89,8.98
DELTA_ANALYZE OUTER_PILLAR_WIDTHS=8.90,8.98,8.99,8.98,8.89,9.04
DELTA_ANALYZE SCALE=1.0
DELTA_ANALYZE CALIBRATE=extended
```

I'll try to dig and find problems myself.

[klippy.log](https://github.com/Klipper3d/klipper/files/8886081/klippy.log)

CODeRUS, 2022-06-12:
found it.
at https://github.com/Klipper3d/klipper/blob/master/klippy/extras/delta_calibrate.py#L48
`obj_angles` should be wrapped to `list()`

Sineos, 2022-06-12:
A PR would be very much appreciated 👍

CODeRUS, 2022-06-12:
PR: https://github.com/Klipper3d/klipper/pull/5569

KevinOConnor, 2022-06-13:
Thanks.  Should be fixed with commit 7f9ea231.

-Kevin

closed 2022-06-13 by CODeRUS