Klipper3d/klipper #5570: Genuine bltouch not working consistently when connected to raspi GPIO

potchin, 2022-06-12:
Following on from a [discussion](https://klipper.discourse.group/t/help-with-bltouch/3113) I started on discourse there looks to be a bug in klipper which is causing inconsistent behaviour when using a bltouch connected to a secondard mcu (raspi) whereby its not reliably detecting the trigger state.

Description of what I have done and full `printer.cfg` are attached to the discourse discussion. I'm happy to run tests or provide additional logs if needed. I am using a genuine bltouch 3.1 and klipper v0.10.0-426 connected to a raspi 3b

Sineos, 2022-06-13:
Please do not cross post. Developers are active on discourse as well and if they have anything to contribute, they will.

closed 2022-06-13 by Sineos

potchin, 2022-06-13:
Bit disappointed in that to be honest - I gave as much info as possible and all the replies I had were basically "it doesnt work on a pi" which, in my opinion, is a bug. I offered to run tests and provide whatever logs were needed but you closed the bug. I'll send my bltouch back and go back to a capacitive sensor, its a shame you're not open to a bit more collaboration to fix what I would expect to become an increasingly common use-case.

Fisheiyy, 2022-06-14:
> Bit disappointed in that to be honest - I gave as much info as possible and all the replies I had were basically "it doesnt work on a pi" which, in my opinion, is a bug. I offered to run tests and [...]

It makes no sense to connect a probe to your pi, it would be a mess of wires and would not work well if at all because the pi was not made for such a task, maybe an arduino that is connected to the pi via usb would be more up to the task but it still makes no sense to do such a thing

potchin, 2022-06-14:
I don't want this to descend into an argument but cable-management shouldn't concern you and the klipper docs explicitly state the benefits of [Multiple Micro-controller Homing and Probing](https://www.klipper3d.org/Multi_MCU_Homing.html#multiple-micro-controller-homing-and-probing) and how to use a [raspi as a secondary mcu](https://www.klipper3d.org/RPi_microcontroller.html#rpi-microcontroller). I'd also point out that connecting essentially a switch to GPIO pins is **exactly** what they are for.

If using a pi (which is what 90% of people are running klipper on) doesn't work in certain cases then these should be clearly documented. Fortunately I was able to return my bltouch but others may not be so lucky and would have likely wasted money in trying to get something working which you deem to "make no sense"