Klipper3d/klipper #5576: filament motion sensor no longer triggers pause when connected to entstop pins

worksasintended, 2022-06-14:
I have this issue with the latest update on 4 different machines running the BTT smart filament sensors successfully before. 
2 Octopus, 1 Octopus Pro, 1 Spider are used as mcu, different klipper installations as well. 
The issue is clearly introduced with an update within the last 4 weeks. None of the merges stood out to me, though. 
No error is visible in the logs. 
Klipper version: v0.10.0-463
Moonraker version: v0.7.1-586-ga8b9fc0

Example configuration:
```
[filament_motion_sensor filament]
detection_length: 10
extruder: extruder
switch_pin: PG10
pause_on_runout: True
 ```

Sineos, 2022-06-14:
As already [stated when you opened the ticket](https://www.klipper3d.org/Contact.html): Issues here are for development only. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

Also post your klippy.log file containing a (simulated) run-out event

worksasintended, 2022-06-14:
This is not a support request, this is a bug report and there is nothing to see in the logs

jakep82, 2022-06-15:
Developers are unlikely to respond here unless a specific commit with a bug has been identified. This needs to be discussed on Discourse to determine the root cause.

closed 2022-06-17 by github-actions[bot]