Klipper3d/klipper #5580: mesh_pps causing excessive load even when idle

dans98, 2022-06-16:
I'm setting up a machine using a RPi Zero 2 W, and I noticed while experimenting with a large bed mesh (probe_count: 29,26), that the klipper process is using an excessive amount of cpu time even when the machine is idle.

Since the mesh takes nearly an hour to make, I altered mesh_y_pps & mesh_x_pps in the saved config section to expedite testing. After a firmware restart, and allowing everything to settle in for few minutes, cpu load on the klipper process looks like the following. 

|PPS|Load|
|------|------|
|0,0 |~9%|
|1,1 |~18%|
|2,2|~23%|
|4,4|~41%|
|8,8|~108%|

Anything higher than 4,4 and klipper will shut down while doing something as simple as a G28, and give the following message.

> MCU 'mcu' shutdown: Timer too close
> This often indicates the host computer is overloaded. Check
> for other processes consuming excessive CPU time, high swap
> usage, disk errors, overheating, unstable voltage, or
> similar system problems on the host computer.

I know at some point it will always overload the pi, but load should not be this high even when sitting idle. 

If i get some time, i'm going to try and dig into the code today, and see what might be going on, as the log doesn't point to anything obvious.

[klippy.log](https://github.com/Klipper3d/klipper/files/8920488/klippy.log)
[image]

dans98, 2022-06-16:
I believe I found the primary problem, but someone will need to explain why this is being done, as I'm not seeing a good reason for it.

https://github.com/Klipper3d/klipper/blob/master/klippy/extras/bed_mesh.py#L218

get_status is being called 4 times a second,, and its returning all the data points for every profile in the system. If a profile is active it data gets sent twice. more importantly the interpolated mesh points are sent for the active mesh.

so with only this mesh in the system.

```
[bed_mesh]
horizontal_move_z: 5 
speed: 300
mesh_min: 5,37
mesh_max: 295, 295
probe_count: 29,26
#mesh_pps: 4, 4
mesh_pps: 0
algorithm: bicubic
move_check_distance: 3
```

58,686 points (if i did the math right) plus some additional "meta" data are being pushed out 4 times a second. 

That's a lot of data to be pushing around in python, is this really needed, and if so why?

Arksine, 2022-06-16:
I don't believe that it is the amount of data being pushed over the socket, as the data is only pushed if there is a change.  However the operation to populate the static dict and check the data on an extremely large mesh could potentially cause an issue, as each point in both matrices are rounded on every call to `get_status()`.  We can and probably should improve this behavior by rounding everything to 6 places when the mesh is generated.

That said I'm unsure if it is necessary (or a good idea) for a front end to subscribe to `bed_mesh`.  While the data isn't entirely static, it can be requested when a user chooses to view the mesh.  I'm unsure of a use case where the front end needs to be notified of a change in the mesh.

I'd also point out that a mesh with points 10-15 mm apart will not likely benefit from additional interpolated points in the mesh.  There isnt going to be a whole lot of z delta between points, so the bicubic algorithm will approximate linear interpolation.

Arksine, 2022-06-17:
I just created a PR that should resolve this.  It would be appreciated if you can checkout the PR and verify that it works as expected, thanks.

dans98, 2022-06-17:
@Arksine I've tested the fix on my machine, and it works very well. CPU load is now 3% to 4% no matter what mesh_y_pps & mesh_x_pps are set to.

As an additional test, I duplicated the default mesh so that I had 15 massive meshes loaded. Other than the initial load spike when swapping between them, load always sat in the  3% to 4% range.

dans98, 2022-06-17:
@Arksine one thing i forgot to ask as an aside, is do you think all the points for the not active profile need to be returned in the status?

I ask, because at least in mainsil, you can't visualize the mesh, unless you load it. Thus sending the points for all the profiles seems like wasted bandwidth. On the other hand, I'm not aware what all the other use cases are.

closed 2022-06-17 by KevinOConnor

qwewer0, 2022-06-17:
@dans98 During printing, did the load change with large mesh after the change?

Arksine, 2022-06-17:
@dans98 Although Mainsail does not use the matrices supplied in the profiles, it would be possible for a front end or application to preview saved profiles.  I don't think bandwidth is a concern as the data would only be sent when its either explicitly requested (or when a profile is added/removed if a front end is subscribed).

@qwewer0 The overall load during a print should decrease if your are using Moonraker with a front end that subscribes to `bed_mesh`, such as Mainsail.  Otherwise its unlikely you will see a change.  Your mesh would also need to be sufficiently large for it to be noticeable.

PR #5584 (2022-06-17): bed_mesh: cache mesh status
Cache the mesh status dict, only updating it when a mesh is loaded, cleared, or if there is a profile change.  This should resolve #5580.