Klipper3d/klipper #5587: BLTouch failed to verify sensor state with dual_carriage enabled (IDEX)

NWSpitfire, 2022-06-18:
Hi,
I have just finished building my custom cartesian IDEX, but I am having problems with my CR-Touch (Creality version of BL-Touch).

The printer homes just fine on X1, X2 & Y, but when T0 moves to deploy the probe in safe Z home position the probe does not deploy and Klipper prints an error "BLTouch failed to verify sensor state". When the printer starts the BLTouch does its usual deploy/retract sequence and when klipper connects it does not report "BLTouch failed to verify sensor state" at startup (unlike if the probe is disconnected). Just on homing with Dual carriage enabled.

I double checked the wiring and it is correct, so I manually told the probe to deploy with probe debug, tested it triggered, and retracted it and all was successful. I then disabled Dual_carriage and the probe worked successfully for home and height map, but when Dual_Carriage was reenabled it would not home again.

As another troubleshooting step I installed an inductive probe and setup as "Probe" and that worked successfully for homing and heightmap with dual_carriage enabled (although somewhat inaccurate - my dodgy probe).

At this point I don't believe there's anything wrong with the hardware (also the probe has worked successfully for the last 4 months on this printer/board and with Dual Carriage disabled). I have included the klippy log and let me know if you want anything else regarding this issue.

Thanks

The current setup is;
MCU; Creality 2.5.2
Auxboard; Creality 1.1.3
Pi; Raspberry Pi 4
IDEX with 2 X-Carriage in a Cartesian Setup

[klippy (4).log](https://github.com/Klipper3d/klipper/files/8933753/klippy.4.log)

Sineos, 2022-06-21:
Github is not used for support requests. See above link. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

Sineos, 2022-06-21:
Unless you can point to the faulty line of code, it is the philosophy here to triage potential bugs in Discourse first. It is very unlikely that you will get any developer's attention here.

closed 2022-06-23 by github-actions[bot]