Klipper3d/klipper #5595: EBB42 V1.1 STM32G0B1:   MCU 'EBBCan' shutdown: Timer too close

cristianku, 2022-06-22:
I am running a Voron 2.4 with Octopus Pro 1.0, BTT U2C 1.1 and EBB42 1.1 STM32G0B1

All are with the latest firmware.

I am running the EBB42 with 4 wires from the U2C, using can bus communication.

I am also running a can bus communication, using a usb cable, from U2C to Octopus pro 1.0

I have set the can bus everywhere to 250000.

and now after printing fine for few minutes, I am getting this error :

16:45:19
MCU 'EBBCan' shutdown: Timer too close

16:45:19

MCU 'EBBCan' shutdown: Timer too close
This often indicates the host computer is overloaded. Check
for other processes consuming excessive CPU time, high swap
usage, disk errors, overheating, unstable voltage, or
similar system problems on the host computer.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

[klippy.log](https://github.com/Klipper3d/klipper/files/8958790/klippy.log)

cristianku, 2022-06-22:
I forgot to add that i am using CanBoot on Klipper Octopus pro 1.0, and this is connected via usb cable to the U2C BTT 1.1

cristianku, 2022-06-22:
I have a raspberry pi 4 2gb

cristianku, 2022-06-23:
Uninstalled Obico ( ex spaghetti ) but still the same problem and always on the same layer ! [image]

cristianku, 2022-06-23:
[image]

cristianku, 2022-06-23:
and if the problem is in EBB42 because the micro steps are too high ( 16 ) ?

cristianku, 2022-06-23:
It's happening always at this layer

[image]

cristianku, 2022-06-23:
maybe thats the reason ?

 Compiling out/src/stm32/can.o
src/stm32/can.c:68:3: warning: #warning CAN on STM32F4 is untested [-Wcpp]
  #warning CAN on STM32F4 is untested
   ^~~~~~~

cristianku, 2022-06-23:
Solution ! the can communication between Bigtreetech U2C and OCTOPUS pro 1.0 STM32F4  is not working !!!!

Sineos, 2022-06-23:
Please shift this discussion to the Klipper Discourse. Matching threads are already existing there:
https://klipper.discourse.group/t/experimental-usb-to-canbus-bridge-mode/3136
https://klipper.discourse.group/t/host-canbus-devices/37/8

closed 2022-06-25 by github-actions[bot]