Klipper3d/klipper #5598: Finding help is hard to do

Cabinfourus, 2022-06-23:
I have always been a hard-core Marlin fan until so many started chatting about this new firmware Klipper. So I had to find out for myself. 
I started by reading over the Klipper website and getting an idea of how it worked. It looked simple enough, so I dove in. 
Two months later I am still battling issues with my configurations. 
Klipper is not for someone who doesn't want to learn to work with codes. It is difficult setting it all up and just when you think you have your printer ready, another issue pops up. 
So here is why I decided to chime in on the Klipper manual. It starts out easy enough but the author(s) forget that some of us are code illiterate. Some of the abbreviated words or instructions can get confusing. It is almost like trying to read the ancient Greek language at times. 
Example: Mesh setup. Trying to figure out how to tell Klipper the size of your print bed and the probe offsets is very confusing. You have used the nozzle as a reference in almost every other setup, like z tile but with a bed mesh, you use the probe. You have to take into consideration the x and y offsets of your probe in order to get the min and max points. 
Another example: Straight from the Klipper manual "The probed coordinate farthest farthest from the origin." Where is the origin? What is the origin? And farthest farthest?  I am lost. 
I also can't seem to find any help with homing my z axis using an NPN probe. 
If someone, anyone, could point me in a direction that I might find "Klipper for Dummies" I would greatly appreciate it. 
This stuff may come easy to some of you but the rest of us are LOST>

jakep82, 2022-06-24:
https://www.klipper3d.org/Contact.html#i-have-a-question-about-klipper

closed 2022-06-26 by github-actions[bot]

LazaroFilm, 2022-08-12:
There are a quite a few places where you can get help easily. 

- [The official discourse forum](https://klipper.discourse.group)
- [Discord group](https://discord.klipper3d.org/)
- [r/Klippers Reddit sub](https://www.reddit.com/r/klippers/)
- [Facebook Klipper group](https://www.facebook.com/groups/klipper)