Klipper3d/klipper #5604: Mega 2560 and Neopixel strip

viappidu, 2022-06-26:
I am having troubles making a 60 leds strip work on extra mcu.
Followed "standard" connection guide with an external power supply (1000uF in place) and a resistor (2x220) on data line.
It works properly if connected to an Octopus Pro (main mcu) but the Mega behavior is erratic changing the result every time.

The Mega is mounted with a 12v powered Ramps 1.6 and I get a very stable 5v on pins.
I already tried changing the Mega for a brand new one and removing the Ramps shield with same (erratic) results

Am I missing something?

ayufan, 2022-06-27:
@viappidu I never got to finish this: https://github.com/Klipper3d/klipper/pull/4525, but Mega 2560 are fairly slow and very often neopixel bitbang is interrupted. Try the mentioned PR

viappidu, 2022-06-28:
Thanks @ayufan!
PR [4525](https://github.com/Klipper3d/klipper/pull/4525) works just fine for me
My bad searching PR for mega only and not for neopixel