Klipper3d/klipper #5624: SET_DISPLAY_TEXT raises "malformed command" error if MSG contains ";"

pedrolamas, 2022-07-03:
SET_DISPLAY_TEXT was introduced in #5593 as a fix for some `M117` override issues, but I've noticed that my M117 override started failing as some of my Slicer plugins add some messages that contain ";"

This works fine:

```
SET_DISPLAY_TEXT MSG="text"
```

But this will fail:

```
SET_DISPLAY_TEXT MSG="text;text"
```

[image]

pedrolamas, 2022-07-04:
Considering the current example on `rawparams` [here](https://www.klipper3d.org/Command_Templates.html#the-rawparams-variable), the easy fix for it seems to be updating the example to cut off anything after ";":

```diff
  [gcode_macro M117]
  rename_existing: M117.1
  gcode:
    {% if rawparams %}
-     {% set escaped_msg = rawparams|replace('"', '\\"') %}
+     {% set escaped_msg = rawparams.split(';')[0]|replace('"', '\\"') %}
      SET_DISPLAY_TEXT MSG="{escaped_msg}"
      RESPOND TYPE=command MSG="{escaped_msg}"
    {% else %}
      SET_DISPLAY_TEXT
    {% endif %}
```

@Arksine as you wrote the original code for this, I thought you would have an opinion on this?

Sonicboom247, 2022-07-17:
i placed a message in quotes and error disappeared, but message does not scroll and is cutoff at 16 char

pedrolamas, 2022-07-17:
@Sonicboom247 check the Pull Request link above for a complete fix for this issue.

Sonicboom247, 2022-07-17:
Thank you, just going to go back to M117 since this did not fix message scrolling and to reduce amount of text on screen to just 4 char.

pedrolamas, 2022-07-19:
Closing as #5625 has been merged with a complete example on how to fix this issue.

closed 2022-07-19 by pedrolamas

goldjunge91, 2022-12-28:
in the example file https://github.com/Klipper3d/klipper/blob/master/config/sample-macros.cfg
```
[gcode_macro M117]
rename_existing: M117.1
gcode:
  {% if rawparams %}
    {% set escaped_msg = rawparams.split(';', 1)[0].split('\x23', 1)[0]|replace('"', '\\"') %}
    SET_DISPLAY_TEXT MSG="{escaped_msg}"
    RESPOND TYPE=command MSG="{escaped_msg}"
  {% else %}
    SET_DISPLAY_TEXT
  {% endif %}
```
But I get this errors:
```
Timelapse: Rendering started
Rendering -
Malformed command 'SET_DISPLAY_TEXT MSG="Rendering \"'
Malformed command 'SET_DISPLAY_TEXT MSG="Rendering \"'
Done printing file
```
If I try again rendering with "TIMELAPSE_RENDER":
```
TIMELAPSE_RENDER
Timelapse: Rendering started
action:paused
Rendering |
Rendering /
Rendering -
Malformed command 'SET_DISPLAY_TEXT MSG="Rendering \"'
Malformed command 'SET_DISPLAY_TEXT MSG="Rendering \"'
```

**from her https://github.com/Klipper3d/klipper/issues/5624#issuecomment-1173514768**
I copy this line 
```
{% set escaped_msg = rawparams.split(';')[0]|replace('"', '\\"') %}
```
and replace this line:
```
{% set escaped_msg = rawparams.split(';', 1)[0].split('\x23', 1)[0]|replace('"', '\\"') %} 
```

And now looks like its working:
```
TIMELAPSE_RENDER
Timelapse: Rendering started
action:paused
Timelapse: Rendering finished
action:resumed
```