Klipper3d/klipper #5637: Extruder motor vibrates but does not move

GitSiam, 2022-07-09:
Hi, i'm using Klipper on SW_X1 withe the original Motherboard and a raspbbery pi 3b+. I change the original extruder for a biqu_h2 (i also experiment some issue withe the original extruder). 

When i try to load, unload or print a file my Extruder stepper vibrates but does not move.

What i have tried :
- change the wiring setup --> fail
- put a brand new nema17 stepper --> Extruder motor vibrates but does not move
- Plug the extruder stepper on the X-axis --> work well

so the problem should comme from the firmware but i cant figure it out.
i check the other Printer.cfg from Github for the same configuration without any result.

do i need to change the motherboard or it can be fix in klipper parameter

Thanks for your help

[klippy.log](https://github.com/Klipper3d/klipper/files/9078079/klippy.log)

Pieter005, 2022-07-10:
hello gitsiam

u may need to buy a new cable, i had the same problem on my i3 clone

Sineos, 2022-07-10:
Github is not used for support / feature requests as already stated when opening the Issue and detailed in above link. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

GitSiam, 2022-07-11:
use github to share the results of work done to improve Klipper. We don't use github for requests.

closed 2022-07-11 by GitSiam