Klipper3d/klipper #5638: Macro with numbers sometimes causes "Unknown Command"

chaydock, 2022-07-10:
Initially I created a macro to retract filament 80mm and named it appropriately "Retract_Filament_80mm". However, when I run the macro it comes up with an error of "Unknown Command: Retract_Filament_80" which is missing the "mm" at the end.

Testing showed that any letters immediately after the numbers were cut off. Even the underscore would be removed if for instance "Test_123_456" was the name of the macro, it would say it couldn't find "Test_123". 

Additional testing showed that if the macro started with a number that the number itself was removed. So, "123_Test" would return an error of unknown "_Test".

The workaround for now is to just be mindful of the behavior and name macros in a way that doesn't cause this issue. So, for my initial macro, I simply renamed it to "Retract_filament_80" and called it a day. All the same, it feels like inappropriate behavior and someone may want to take a poke at why it does that.

Sineos, 2022-07-10:
Github is not used for support / feature requests as already stated when opening the Issue and detailed in above link. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

Also: https://www.klipper3d.org/Command_Templates.html#g-code-macro-naming

chaydock, 2022-07-10:
This is not a support/feature request. It's a bug report... Or at least I perceive it as one since the behavior exhibited does not appear to be correct/desirable. Granted, this is not a 5-alarm fire bug... Far from it... But a bug all the same. If this is not the correct place to report such an issue, I'm happy to use another mechanism. This by my account appeared to be the proper channel for such reporting. But, if not, please let me know the one you prefer.

chaydock, 2022-07-10:
@Sineos ... I'm looking at your reply on my PC now (I looked at it on my phone earlier)... The link you provided for the macro naming I guess makes sense that the names I tried using and testing with previously failed. And... I can understand your reply thinking that this is a feature request to change/fix this behavior. I was unaware of the number requirement listed. I must admit I find it rather odd as a naming requirement IMHO... but it is what it is. 

Anyway... Given that the names are invalid based on the requirements listed in the link... should this type of issue not be registered as a configuration failure at restart? A configuration validation check? Similar to for instance when you fail to include the microsteps setting in a stepper config? Upon reading the file, should it not validate that the macros are using valid names and prompt the user to fix this discrepancy? I'm not sure if one would perceive that also as a feature request. Furthermore, I'm not sure what would be easier... adding a name validation test or allowing numbers to be at any location in a name rather than just at the very end. Although... I'm sure people will find other creative ways to break macro names. ;)