Klipper3d/klipper #5645: UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)

cfelicio, 2022-07-14:
I'm using Klipper with Fluidd and it works fine (everything prints out), but at the very end of a print with the Palette 2 in connected mode, the MCU shuts down with this error:

Unhandled exception during run
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 217, in run
    self.reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 292, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 340, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/pi/klipper/klippy/reactor.py", line 158, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/pi/klipper/klippy/extras/palette2.py", line 547, in _run_Read
    text_buffer = self.read_buffer + str(raw_bytes.decode())
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)
Transition to shutdown state: Unhandled exception during run
Dumping gcode input 0 blocks

See attached files for full log. Let me know if any questions. Thanks for your help! :-)

[201901MosaicKeychain1.p2pp.zip](https://github.com/Klipper3d/klipper/files/9108627/201901MosaicKeychain1.p2pp.zip)
[klippy.log](https://github.com/Klipper3d/klipper/files/9108624/klippy.log)

cfelicio, 2022-07-14:
For reference: https://github.com/Klipper3d/klipper/pull/4057

Sineos, 2022-07-14:
Ping @cmroche

Sonnenwerk, 2022-08-02:
I had this issue today. It is caused e.g. by a german special sign "ü" or "ä", which is in the filename.
I used prusaslicer 2.4 and it was reproducable.

cmroche, 2022-08-06:
Sorry for the delay folks, I'll take a look at the error and see if I can repro, or at least discard it following what @Sonnenwerk mentioned.

Sineos, 2022-08-19:
#5694

ReXT3D, 2022-08-19:
I'm not sure if https://github.com/Klipper3d/klipper/pull/5694 was also supposed to address the case that I encountered.  Issuing pretty quotes in place of regular double quotes still causes Klipper shutdown:

[image]

[image]

Thank you.

Sineos, 2022-08-19:
> I'm not sure if #5694 was also supposed to address the case that I encountered. Issuing pretty quotes in place of regular double quotes still causes Klipper shutdown:

I think this is a completely different issue. Pretty quotes IMO were never meant to work in a Klipper command. This is like you would subsitute quotes by pretty quotes in C or JS etc and expect the compiler / runtime to like it.

ReXT3D, 2022-08-19:
> I think this is a completely different issue. Pretty quotes IMO were never meant to work in a Klipper command. This is like you would subsitute quotes by pretty quotes in C or JS etc and expect the compiler / runtime to like it.

I understand and I agree.  I guess my only point here is that it would be nice to have a bit more graceful error handling than a machine shutdown.  In case someone goofs typing a console command 40 hours into a print, let's say...

closed 2022-08-27 by github-actions[bot]