Klipper3d/klipper #5646: how/where to connect hall filament width sensor on octopus v1.1

seebaer1976, 2022-07-17:
Hi,
got the octopus v1.1.

now I have recreated the hall filament width sensor([link to hall filament width sensor](https://www.klipper3d.org/Hall_Filament_Width_Sensor.html)) and would like to connect it.
unfortunately I can't find anything to which pins I have to/can connect it.

does anyone have an idea?

Kind regards
mark

JamesH1978, 2022-07-18:
I believe you needs a couple of pins capable of analog input with that sensor. I assume you could use a couple of the thermistor ports. But this is not a GitHub issue, this is used for triaged bugs to be fixed, not general help, it is better discussed on the discourse or the discord servers. 

Thanks
James

seebaer1976, 2022-07-18:
OK thanks
my mistake

closed 2022-07-18 by seebaer1976