Klipper3d/klipper #5648: Adding official support for ADXL343 and ADXL346

Max-Roesenberger, 2022-07-19:
the ADXL343 is already unofficially supported due to having the same device ID as the ADXL345. 
the ADXL346 has the same adresses and general functionality as the ADXL345 and is a drop in replacement exept for the different device ID. I have bodged the ADXL345.py on my printer to accept the different device ID and the ADXL346 is working flawlessly.
I attached my klippy.log and my bodged python code. The bodge is adding the id of the ADXL346 in line 23 and making the code accept that id in line 375.

[klippy.log](https://github.com/Klipper3d/klipper/files/9140506/klippy.log)
[adxl345.py.txt](https://github.com/Klipper3d/klipper/files/9140512/adxl345.py.txt)

Sineos, 2022-07-19:
There seem to be some differences between the 345 and 346. See https://ez.analog.com/mems/f/q-a/81958/adxl345-adxl346
Don't know if it has an effect on the measurement.
Ping @dmbutyugin

Max-Roesenberger, 2022-07-19:
From my understanding its about the least significant bit being 0 since it falls outside the valid data range. so the 0 is just a defined state.

closed 2022-08-31 by github-actions[bot]