Klipper3d/klipper #5664: Question about SHT2x series sensor

acha666, 2022-07-27:
It looks like that klipper support SHT21 IIC temperature sensor, however I get error `index out of range`

I'm working on a[ Toolhead PCB](https://github.com/acha666/RP2040_ToolheadPCB_SB) based on RP2040 microcontroller, which use a SHT21 sensor for measuring chamber temperature

After checking [code](https://github.com/Klipper3d/klipper/blob/master/klippy/extras/htu21d.py) , [SHT21 Datasheet](https://cdn.sensirion.com.cn/media/documents/120BBE4C/61642236/Sensirion_Humidity_Sensors_SHT21_Datasheet.pdf) and [SI7021 Datasheet](https://www.silabs.com/documents/public/data-sheets/Si7021-A20.pdf)

It seems that klipper try to get the `device id` of the sensor, by using command `0xFA 0x0F 0xFC 0xC9`
[image]
[image]

This command does mentioned in SI7021 Datasheet
[image]

But I can't find this command in the datasheet of SHT21
[image]

It seems that SHT21 sensor doesn't have a `Electronic Serial Number` like SI7021, so trying to read the device ID of SHT21 cause the error

Here is my log 
[klipper log.txt](https://github.com/Klipper3d/klipper/files/9200863/klipper.log.txt)

acha666, 2022-07-27:
**Chinese Version**

Klipper文档中提到klipper支持SHT21 IIC温度传感器，但是我遇到报错`index out of range`

我自己绘制了一块基于RP2040的[工具头板](https://github.com/acha666/RP2040_ToolheadPCB_SB)，使用SHT21作为箱温测量

在查看了 [相关代码](https://github.com/Klipper3d/klipper/blob/master/klippy/extras/htu21d.py) , [SHT21 Datasheet](https://cdn.sensirion.com.cn/media/documents/120BBE4C/61642236/Sensirion_Humidity_Sensors_SHT21_Datasheet.pdf) 和 [SI7021 Datasheet](https://www.silabs.com/documents/public/data-sheets/Si7021-A20.pdf) 之后

看起来klipper在尝试使用命令`0xFA 0x0F 0xFC 0xC9`获取传感器的`device id`
[image]
[image]

这一命令在SI7021的Datasheet中有被提到
[image]

但是在SHT21的Datasheet中找不到这一命令
[image]

看起来SHT21并不像SI7021一样，有一个 `Electronic Serial Number` 寄存器，所以尝试获取SHT21的`device id`时产生了这个错误。

不知道我的猜测是否正确，恳请各位大佬帮忙确认一下问题

Klipper日志： [klipper log.txt](https://github.com/Klipper3d/klipper/files/9200863/klipper.log.txt)

Sineos, 2022-07-31:
Github is not used for support / feature requests as already stated when opening the Issue and detailed in above link. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

Also note that the SHT21 sensor is marked as "untested" here: https://github.com/Klipper3d/klipper/blob/38e82e8d8f4d27edbefaa822d8ecce229c4ce8c5/klippy/extras/htu21d.py#L18
It is possible that it will require some development work to get it working.

closed 2022-08-02 by github-actions[bot]