Klipper3d/klipper #5671: Invalid speed in 'G0 F0 X91.354 Y115.457 Z0.4'

mustytechy, 2022-07-31:
great mate , i run in to error please after installation and i tried to print calibration cude but  i also get this error Invalid speed in 'G0 F0 X101.5 Y117.79 Z0.4' , please how can i fix it  and thanks in advance.

Afterschnitzel, 2022-07-31:
This is because you try to run stepper with feedrate F0

https://github.com/Klipper3d/klipper/issues/173

mustytechy, 2022-07-31:
I understand, but do you have any idea how to fix this please

Sineos, 2022-07-31:
Github is not used for support / feature requests as already stated when opening the Issue and detailed in above link. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

Afterschnitzel, 2022-07-31:
It depends what slicer you use.

closed 2022-08-02 by github-actions[bot]