Klipper3d/klipper #5680: Delta dual extruder XY offset

Math1010, 2022-08-01:
Hello,
I have a custom delta printer with two extruder and a probe.
Each extruder has an XY offset regarding to the bed center origin : 
[image]

I try to declare XY offset like this :

```
[gcode_macro T0]
gcode:
    SET_GCODE_OFFSET X=-7.5                     # X offset
    SET_GCODE_OFFSET Y=-12.99                    # X offset
    ACTIVATE_EXTRUDER EXTRUDER=extruder

[gcode_macro T1]
gcode:
    SET_GCODE_OFFSET X=-7.5                     # X offset
    SET_GCODE_OFFSET Y=-12.99                    # X offset
    ACTIVATE_EXTRUDER EXTRUDER=extruder1
```
But after homing if I command X0 Y0 move, the active extruder is not a the center of the bed.
How can I set XY offset for dual extruder on a delta printer ? 

Thanks

Sineos, 2022-08-01:
Github is not used for support / feature requests as already stated when opening the Issue and detailed in above link. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

Math1010, 2022-08-01:
Created a support request :
https://klipper.discourse.group/t/delta-dual-extruder-xy-offset/3668

closed 2022-08-01 by Math1010