Klipper3d/klipper #5681: [Feature request] Missed dual_carriage status reference on cartesian printer...

themacboy, 2022-08-01:
Good morning,

I have created a basic IDEX Cartesian printer and all the coding works fine with the double carriage handling.

But for some reason I tried to get response from printer.dual_carriage.active_carriage and got the following error:

´´´
Error evaluating "gcode_macro CONFIGS_IDEX:gcode": UndefinedError: "instance object" does not have attribute "dual_carriage"
´´´

After checking the project documentation I read:

´´´
The following information is available in [dual_carriage] on a hybrid_corexy or hybrid_corexz robot

´´´

Then I understood why I'm getting the error and managed a solution in a different way.

But honestly, I'm not sure I understand this limitation (I'm sure there's a good reason hehehe).

I'll suggest adding cartesian to the list (I'm not good at coding Python or C to do this on my own, sorry)

Sineos, 2022-08-01:
Github is not used for support / feature requests as already stated when opening the Issue and detailed in above link. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

themacboy, 2022-08-01:
> Github is not used for support / feature requests as already stated when opening the Issue and detailed in above link. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

Oh ! Sorry, looking at the problem I thought that was a enhancement of a current feature (not new) or better a bug int he current design. 

Thx

themacboy, 2022-08-01:
Moved to:

https://klipper.discourse.group/t/enhancement-missed-dual-carriage-status-reference-on-cartesian-printer/3673