Klipper3d/klipper #5685: BLtouch clone fails to maintain state, breaking stow/pin_up

four0four, 2022-08-02:
I'm not sure what the provenance of my particular clone is, but it's got a nasty behavior of deploying when there's no PWM signal. In Marlin, this isn't an issue: the Arduino-ish libraries underpinning most controllers appear to just set-and-forget a toggle-on-compare peripheral. 

In Klipper, `extras/bltouch.py` disables PWM  after a short period. This results in the clone deploying its probe immediately following any state change ie: `pin_up` is immediately followed by a deployed pin. G28 mostly works, but any moves with Z <= z_offset result in the probe colliding with the bed.

I fixed it locally by adding a config item to disable the second call to `mcu_pwm.set_pwm` here: https://github.com/Klipper3d/klipper/blob/a709ba43af8edaaa307775ed73cb49fac2b5e550/klippy/extras/bltouch.py#L110

```
diff --git a/klippy/extras/bltouch.py b/klippy/extras/bltouch.py
index 48759752..de25de96 100644
--- a/klippy/extras/bltouch.py
+++ b/klippy/extras/bltouch.py
@@ -35,6 +35,7 @@ class BLTouchEndstopWrapper:
                                                      True)
         self.probe_touch_mode = config.getboolean('probe_with_touch_mode',
                                                   False)
+        self.pwm_hack = config.getboolean("pwm_hack", False)
         # Create a pwm object to handle the control pin
         ppins = self.printer.lookup_object('pins')
         self.mcu_pwm = ppins.setup_pin('pwm', config.get('control_pin'))
@@ -107,7 +108,8 @@ class BLTouchEndstopWrapper:
         end_time = mcu.clock_to_print_time(cmd_clock)
         # Schedule command followed by PWM disable
         self.mcu_pwm.set_pwm(self.next_cmd_time, Commands[cmd] / SIGNAL_PERIOD)
-        self.mcu_pwm.set_pwm(end_time, 0.)
+        if not self.pwm_hack:
+            self.mcu_pwm.set_pwm(end_time, 0.)
         # Update time tracking
         self.action_end_time = self.next_cmd_time + duration
         self.next_cmd_time = max(self.action_end_time, end_time + MIN_CMD_TIME)

```

 I poked around looking for other instances of this issue, but wasn't able to find any. This is definitely a bit of a hack though, I'm not familiar with the codebase!

Sineos, 2022-08-02:
Sounds like https://github.com/Klipper3d/klipper/pull/5088 
Also see https://github.com/Klipper3d/klipper/issues/5080#issuecomment-1003173443

four0four, 2022-08-02:
I think not quite? I might be missing something, but I have always had `probe_with_touch_mode` disabled. 

See in your second link:

"The problem comes in when you try to follow the docs for testing it. `BLTOUCH_DEBUG COMMAND=pin_down` and `BLTOUCH_DEBUG COMMAND=pin_up `should work. "

These don't work, because the pin drops as soon as the timer for sending the `pin_up` pulsewidth expires.

Sineos, 2022-08-03:
Never heard about such an issue.
It makes sense to place this on [Klipper Discourse](https://klipper.discourse.group/) to see if more users pop up with this issue.

closed 2022-08-05 by github-actions[bot]