Klipper3d/klipper #5688: klipper/config/printer-creality-ender3-v2-2020.cfg : FLIP FLOPPED SETTINGS

Cluck0matiC, 2022-08-03:
step_pin: PC2
dir_pin: PB9

I believe these are reversed. My Ender 3 V2 wouldn't move until I compared to someones working config. This was the case for all the steppers. 

Just an FYI.

Sineos, 2022-08-03:
Comparing to Marlin and assuming you have a Creality 4.2.2 board, the information should be correct: https://github.com/MarlinFirmware/Marlin/blob/35d9920ef8ce3b0b73a88df3d3923fbc3b9d3d23/Marlin/src/pins/stm32f1/pins_CREALITY_V4.h#L104-L108

```
[stepper_x]
step_pin: PC2
dir_pin: PB9
```

If there is need to discuss this further, I suggest you open a topic on [Klipper Discourse](https://klipper.discourse.group/)

four0four, 2022-08-03:
It looks like Creality made this change between the 4.2.2 hw rev and 4.2.7, ref: https://github.com/MarlinFirmware/Marlin/blob/35d9920ef8ce3b0b73a88df3d3923fbc3b9d3d23/Marlin/src/pins/stm32f1/pins_CREALITY_V427.h#L34-L35

It might be useful to track board revisions separately a la Marlin here, or at least include some indication of the known differences.

closed 2022-08-05 by github-actions[bot]