Klipper3d/klipper #5695: SCREWS TILT ADJUST bed doesn't lower before probing

FlatEarthDave, 2022-08-08:
The bed does not lower after the first probe of the back left, and any subsequent probes where the corner is higher than the previous probe will cause the probe to not deploy completely and throws an error stating the probe did not deploy.

Sineos, 2022-08-08:
Github is not used for support / feature requests as already stated when opening the Issue and detailed in above link. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

closed 2022-08-10 by github-actions[bot]

FlatEarthDave, 2022-08-11:
It wasn’t for support. I don’t need support, because I fixed the issue myself. I added spacers to my bltouch. It is an overlooked flaw that should be changed in the firmware, so if the user has less than 2mm of distance between the bltouch and the nozzle, it would still work for them too. Sent from Mail for Windows From: SineosSent: Monday, August 8, 2022 4:00 AMTo: Klipper3d/klipperCc: FlatEarthDave; AuthorSubject: Re: [Klipper3d/klipper] SCREWS TILT ADJUST bed doesn't lower before probing (Issue #5695) Github is not used for support / feature requests as already stated when opening the Issue and detailed in above link. Please be so kind to place your support / feature request on Klipper Discourse