Klipper3d/klipper #5700: Klipper is very unstable

pinggi, 2022-08-09:
I use OctoPrint 1.8.1 + OctoKlipper  + Klipper on Raspberry PI connected via USB to RAMPS with atmega (fw v0.10.0-546-ga709ba43).

Every time when I plug in the printer (raspi and ramps) into the outlet. I have to restart Firmware & Host several times to be able to get it 'ready' and use it.

[image]

That is something I can live with, but what is insane is that I have to home XY and Z before before sending move commands otherwise it disconnects on error.
For example, I can't move the hotend up 1 cm from the middle - I have to home it first - wait for it till it goes down 10s near the bed and then another 10s back to be able to just move the hot end 1 cm up which is about 1s. This is crazy.

Now it happened that I wanted to replace the nozzle, but I accidentally cut the wire of the temperature sensor.
Guess what,  I am no longer able to move with the XYZ axes because Klipper is unable to execute commands and disconnects from RAMPS!

Error: ADC out of range.

[image]

Why does the error interrupt printer control and all functions?

The fact that Klipper keeps disconnecting on any error makes it useless and it's insane.

To make Klipper usable, make it fault-tolerant.

Sineos, 2022-08-09:
Github is not used for support / feature requests as already stated when opening the Issue and detailed in above link. Please be so kind to place your support / feature request on [Klipper Discourse](https://klipper.discourse.group/)

Also attach your klippy.log there

pinggi, 2022-08-09:
Yeah, I understand. All right.

closed 2022-08-11 by github-actions[bot]