Klipper3d/klipper #5706: `make menuconfig` not works if the parent directory contains spaces

sl1pkn07, 2022-08-11:
if the parent directory contains spaces, `make menuconfig` fails

~~~
┌─┤[$]|[sl1pkn07]|[sL1pKn07]|[/media/COSAS/aplicaciones/Creality Ender 3/klipper]|
└───╼  make menuconfig
No change to configuration in '/media/COSAS/aplicaciones/Creality Ender 3/klipper/.config'
Loaded configuration '/media/COSAS/aplicaciones/Creality Ender 3/klipper/.config'
---snip----
~~~

but if move the main directory to one without spaces:

~~~
┌─┤[$]|[sl1pkn07]|[sL1pKn07]|[/media/COSAS/aplicaciones/klipper]|
└───╼  make menuconfig
(Top)
                                                                                                                                                                 Klipper Firmware Configuration
[ ] Enable extra low-level configuration options
    Micro-controller Architecture (Atmega AVR)  --->
    Processor model (atmega2560)  --->

[Space/Enter] Toggle/enter      [?] Help            [/] Search
[Q] Quit (prompts for save)     [ESC] Leave menu
~~~

start works

greetings

sl1pkn07, 2022-09-16:
ping

sl1pkn07, 2022-10-22:
ping

Neko-vecter, 2022-11-09:
The best solution probably use “-” to replace space.

Sineos, 2022-11-10:
Send a PR?

sl1pkn07, 2022-11-10:
seems involved python scripts, so no, i cant. sorry

greetings

sl1pkn07, 2022-12-16:
ping

Neko-vecter, 2022-12-16:
I think this error is caused by menuconfig instead of klipper. 

More info in https://wiki.st.com/stm32mpu/wiki/Menuconfig_or_how_to_configure_kernel

djessup, 2023-01-20:
Make and menuconfig both struggle with spaces. This is not a Klipper problem IMO.

sl1pkn07, 2023-02-24:
ping

Neko-vecter, 2023-02-28:
I checked the section on `make menuconfig` configuration. This is part of [Kconfiglib](https://github.com/ulfalizer/Kconfiglib) library. 
Which means you need to go to [Kconfiglib](https://github.com/ulfalizer/Kconfiglib) and open an issue.

sl1pkn07, 2023-02-28:
https://github.com/ulfalizer/Kconfiglib/issues/125

done

sl1pkn07, 2023-04-04:
ping again

closed 2023-05-26 by github-actions[bot]