Klipper3d/klipper #5716: MCU 'mcu' shutdown: Scheduled digital out event will exceed max_duration

Flyguy62, 2022-08-18:
I have been having this issue for a while now. it happens every time the macro for clog or runout is run by the ERCF scripts. basically without fail. I have worked with ETTE the designer of the ercf and tried changing my macros. but nothing has really worked. i am thinking at this point that it could be a klipper issue.
[image]
[klippy (69).log](https://github.com/Klipper3d/klipper/files/9375002/klippy.69.log)

Flyguy62, 2022-08-18:
also as an update it only causes the shutdown when the printer automatically detects runout or clog, when i call the script manually it completes just fine. heres the log file for that
[klippy (70).log](https://github.com/Klipper3d/klipper/files/9375201/klippy.70.log)
.

Sineos, 2022-08-18:
As already noted when you opened this topic: GitHub shall only be used for development purposes. Please be so kind to open such support topics on [klipper.discourse.group](https://klipper.discourse.group/) and attach your klippy.log file there.

closed 2022-08-20 by github-actions[bot]